PX4/PX4-Devguide #25: Add docs for best logging practices

LorenzMeier, 2016-07-31:
@bkueng Could you create a page or section for the SD card recommendations and a brief explanation why this matters (500 ms allowed block time per SD card standard and associated buffering needs?)

I think that would be very valuable for developers.

bkueng, 2016-08-02:
Did you see this: http://dev.px4.io/advanced-logging.html#dropouts ?
Should I write more?

LorenzMeier, 2016-08-02:
Yes, we should recommend individual models and list their performance.

kd0aij, 2016-08-02:
don't forget this data: https://github.com/PX4/Firmware/issues/4634

bkueng, 2016-08-02:
@kd0aij Yes I took the data from there and summed it up in the dev-guide.

closed 2016-08-02 by LorenzMeier

kd0aij, 2016-08-02:
@bkueng I think we want the absolute max for the last column in the table, not an average.
That SanDisk Extreme U3 32GB card must be doing its own buffering, since the variance in the max block write time is so small.

@pkocmoud Is this the same card? https://www.amazon.com/dp/B013CP3KZ6/ref=psdc_1197396_t1_B00MBFPT44

If so, I'll order some of the 16 and 32 GB versions and try them myself.

bkueng, 2016-08-02:
@kd0aij I wanted to have some statistical meaningful value, so I took the average over the spikes. It should give the reader a good idea to compare the cards, not a once-in-a-lifetime value that is potentially never reached again. The link to the more precise measurements is still there as well.
But I don't feel too strong about this, so feel free to edit.

pkocmoud, 2016-08-02:
@kd0aij I have none of these cards.

kd0aij, 2016-08-02:
@bkueng My thinking was that we want to know "worst-case" latency in order to determine the buffer size required to prevent dropouts. The max for the best card is only 17msec with an avg throughput of 460KB/sec, so the buffer size required for that speed would be less than 8000 bytes.

But I'm more curious about why a card advertised as having a 40MB/sec write speed is achieving less than 0.5MB/sec in our tests.

@pkocmoud Which SanDisk Extreme U3 32GB card did you test?

bkueng, 2016-08-02:
yeah that's a good argument as well. I was wondering about write speed too, so I tested under Linux, and the same cards actually achieved much higher throughput, around 10x what I measured with Pixracer. Not sure why though.
I tested the Extreme U3.

kd0aij, 2016-08-02:
@pkocmoud  Sorry, I didn't notice that wasn't you testing the U3

kd0aij, 2016-08-07:
@bkueng I tested a Sandisk Extreme U3 16GB card, and the first run results were similar to your 32GB card test, with max write times of about 17msec. But after I noticed dropouts at about 50KB/sec with the new logger on master, I reran sd_bench, and most of the max write times are over 250msec. Reformatting the SD card had no effect, max write times are still > 250msec.

Could you run a test writing an actual log file using master? I'm curious as to whether my 16GB card is behaving very differently from your 32GB card.

bkueng, 2016-08-08:
@kd0aij That seems weird. I've been using my U3 32GB quite often with the new logger, including today with latest master, and no dropouts. Your linked card is the same, appart from the size. Do you have a 32GB version too?

kd0aij, 2016-08-08:
No, will try it in other vehicles

PR #26 (2016-08-02): Logging: add SD Cards performance results
fixes #25