PX4/PX4-Devguide #46: Mavros gazebo envirment

kiti1013, 2016-10-02:
I try to run mavros with gazebo, but, i did not find the file  integrationtests/setup_gazebo_ros.bash  in the firmware, i am a green hand in ros

eric1221bday, 2016-10-04:
They have significantly altered the process of running gazebo in ros. If you're using the newest master branch then it's easier for you to clone their site_gazebo repo separately and build both that and the firmware separately as ROS packages. Then when you source the setup.bash that would be generated by the catkin build you should be able to launch normally, check all the launch files under Firmware/launch

julianoes, 2016-10-11:
@kiti1013 did you have any success with this?

julianoes, 2016-10-15:
@kiti1013 you are right. The [documentation](http://dev.px4.io/simulation-ros-interface.html) is outdated and needs fixing.

@AndreasAntener and @vilhjalmur89 can you help please?

eric1221bday, 2016-10-15:
Speaking of this even though the sitl_gazebo package is cloned along with the firmware, when trying to build the px4 package catkin doesn't seem able to detect the sitl_gazebo package, necessitating the workaround I mentioned above, is there a way to fix this?

julianoes, 2016-10-17:
@kiti1013 does #54 work for you?

closed 2016-10-17 by LorenzMeier

csorice147, 2016-10-22:
@eric1221bday what is this sitl_gazebo repo?

eric1221bday, 2016-10-22:
@csorice147 they updated their devguide so for now see if they're suggested method works, but if it doesn't, the sitl-gazebo repo is https://github.com/PX4/sitl_gazebo, just clone it and build it alongside the px4 ros package

PR #54 (2016-10-17): simulation: update ROS gazebo instructions
Fixes #46.