PX4/PX4-Devguide #61: Error in documentation for building code for snapdragon

sharvashish, 2016-10-21:
In this [link](dev.px4.io/starting-building.html)
Under Run it for Snapdragon Flight,

to launch mini-dm, the command should be:
**${HEXAGON_SDK_ROOT}/tools/debug/mini-dm/Linux_Debug/mini-dm**

instead of **${HEXAGON_SDK_ROOT}/tools/mini-dm/Linux_Debug/mini-dm**

julianoes, 2016-10-21:
@sharvashish thanks for reporting. Fixed in #62.

closed 2016-10-21 by LorenzMeier

sharvashish, 2016-10-22:
Hi, guys sorry for the wrong link,
But I was trying to mention this link: [here](http://dev.px4.io/starting-building.html#run-it)

Even though raising this issue corrected the same error in few other places, the error is repeated here as well. Please correct it here too. 

Cheers.

julianoes, 2016-10-23:
@sharvashish oh yes there as yet another one, thanks.
Fixed in #63.

PR #62 (2016-10-21): Fix mini-dm path, add note about nano-dm
The path to mini-dm was wrong (fixes #61).

Also, for Mac, there is an alternative with [nano-dm](https://github.com/kevinmehall/nano-dm).