PX4/PX4-Devguide #66: Gazebo ROS interface documentation error

eric1221bday, 2016-11-23:
As it currently stands the documentation on interface to ROS with gazebo is not quite correct. 

First, the second instruction on launching gazebo with ROS wrappers
```
make posix_sitl_default
```
would not compile the sitl_gazebo submodule. If the user has not previously built the sitl_gazebo package, the instruction should be
```
make posix_sitl_default gazebo
```
which would force the makefile to build the sitl_gazebo package alongside

Second, even if the above correction is applied, the simulation still would not run properly if the 
```
mavros_posix_sitl.launch
```
file is run, specifically mavros would be unable to connect to the simulated drone in gazebo.

I've personally found a workaround to this problem and is currently able to run mavros and gazebo in both Ubuntu 14.04 and 16.04 using Gazebo 6 and 7. It is as follows

1. create a catkin workspace in a folder
```
mkdir catkin_ws && cd catkin_ws
catkin init 
mkdir src
```
2. clone both the firmware and the sitl_gazebo package into the src folder of the workspace
```
cd src
git clone https://github.com/PX4/Firmware.git
git clone https://github.com/PX4/sitl_gazebo.git
cd Firmware
git submodule update --init --recursive
cd ../..
```
3. build the catkin workspace with both packages, source the workspace, and run
```
catkin build
source devel/setup.bash
roslaunch px4 mavros_posix_sitl.launch
```

I realize that the sitl_gazebo package is contained within the Firmware folder as a submodule, and thus this methods is quite inefficient in terms of space usage, however this remains the only method I know of that could run mavros with gazebo, and as such I would like the ask the devs if this can be the recommended method on the devguide.

If so, I would happily submit a pull request to update the devguide, if not, then this can be a reference to others as a workaround for the current system

julianoes, 2016-11-23:
@eric1221bday Thanks for the elaborate feedback. This is appreciated.

@vilhjalmur89 you also ran into this issue with the redundant gazebo copies. Can you comment?

@eric1221bday it would be great if you can file a pull request for the things which are obviously wrong and we should try to fix the confusion about the build of sitl_gazebo.

eric1221bday, 2016-11-23:
@julianoes so just as a warning large parts of that page is basically invalid, the "what's happening behind the scenes" section is pretty much completely obsolete. I guess what I'm trying to say that the page would probably need an overhaul, the only section that as it stand works is probably just the top section, launch mavros to interface with no gazebo plugins

this first pull request would probably just be focussed on correcting the versioning on installing gazebo, I'll leave modifying the launch procedure for gazebo wrappers up for further discussion

eric1221bday, 2016-11-23:
@julianoes pull request created

vilhjalmur89, 2016-11-24:
@eric1221bday Thanks for the feedback.

You are right about having to run `make posix_sitl_default gazebo`.

The sitl_gazebo issue is because of the line
`source Tools/setup_gazebo.bash $(pwd) build_posix_sitl_default`,
which should be 
`source Tools/setup_gazebo.bash $(pwd) $(pwd)/build_posix_sitl_default`

julianoes, 2016-11-24:
@vilhjalmur89 can you please review the PR, thanks.

eric1221bday, 2016-11-24:
@vilhjalmur89 you are correct, I am able to run SITL now using the corrected version of the instructions, however I would also advise removing 
```
source ~/catkin_ws/devel/setup.bash
```

since having a catkin workspace is unnecessary to running this SITL, only if the user has some external packages he or she needs to run, that would be up to them, this line just causes confusion.

vilhjalmur89, 2016-11-25:
You might need that line for the Mavros example, but it wouldn't hurt to comment on that

eric1221bday, 2016-11-25:
@vilhjalmur89 actually since there are debian packages of mavros, if you didn't build mavros from source you actually wouldn't need that line at all

vilhjalmur89, 2016-11-25:
@eric1221bday Yes, but in the guide it is recommended to build it from source.

But you're right, it is unnecessarily confusing

eric1221bday, 2016-11-25:
@vilhjalmur89 alright then on second pass I'll leave it in but leave a note on the fact that it's optional

LorenzMeier, 2017-06-15:
@vilhjalmur89 Can you review and close or fix? Thanks!

closed 2017-06-19 by vilhjalmur89

rahulkalliadan, 2018-06-25:
Hello,

I have tried working on the code you provided. 
When I tried Catkin Build, only  1 of 3 packages got succeeded.

[image]

julianoes, 2018-06-25:
@rahulkalliadan this is an older issues, please create a new issue for your problem and copy&paste the error message instead of a screenshot.