PX4/PX4-Devguide #90: devguide broken

Stifael, 2017-03-01:
currently it is not possible to merge due to broken tree.

simonegu, 2017-03-03:
Webhook events `push` fails here is an example of a response:
```
{"ok":false,"event":"push","error":"Can not fast forward from 2efc1496adf4c590a67b75304d149749afd595db to 14c59eea327c970c9387a80163c3b5784d67cc57 (refs/heads/master - refs/heads/master)"}
```

dagar, 2017-03-03:
My guess is we'll need to manually merge and force update either the gitbook repo (https://git.gitbook.com/px4/devguide.git) or the github repo (https://git.gitbook.com/px4/devguide.git) then someone with appropriate access to the gitbook needs to force the sync in the appropriate direction.

Going forward using the Gitbook web editor should be preferred, and we need to make sure the gitbook sync is fixed early if it breaks again. It seems quite flaky.

bkueng, 2017-03-14:
Finally fixed. It was actually quite easy: just merge the gitbook remote master (https://git.gitbook.com/px4/devguide.git) into the github repo & push.

closed 2017-03-14 by bkueng