PX4/PX4-Devguide #92: Richquotes - notes, warnings, tips etc currently not working

hamishwillee, 2017-03-07:
Plugins for heading anchors and richquotes are in conflict: 
https://github.com/erixtekila/gitbook-plugin-richquotes/issues/25

This is just for tracking.

hamishwillee, 2017-04-10:
@bkueng I also note that in [some places](https://raw.githubusercontent.com/PX4/Devguide/master/hardware-snapdragon.md) we use explicit div markup like this
```
<aside class="warning">
Although the Snapdragon uses DF13 connectors, the pinout is different from Pixhawk.
</aside>
```
Can I assume that this was the old way of doing asides, and now the intention is to use richquotes plugin?

bkueng, 2017-04-10:
> Can I assume that this was the old way of doing asides, and now the intention is to use richquotes plugin?

yes

hamishwillee, 2017-04-11:
Thanks @bkueng - I've created #117 to fix up all the incorrect cases. Merge if they look OK!

hamishwillee, 2017-05-10:
@bkueng I think we should disable the heading anchors plugin. This will enable the notes, tips etc to work. 

While it would be nice to have the little link showing the anchor (for easy copying) I think that is not important (note that the actual anchor is still created by the page toc plugin)

What do you think?

bkueng, 2017-05-10:
I agree it would be nice to have it, but if you can still get the link via TOC, then yeah, let's disable it.

hamishwillee, 2017-05-11:
OK, done. We should retest anchors in future though. It is a cool plugin.

hamishwillee, 2018-12-03:
Fixed by https://github.com/PX4/Devguide/pull/680https://github.com/PX4/Devguide/pull/680

closed 2018-12-03 by hamishwillee