PX4/PX4-Devguide #93: Potentially incorrect instructions for Snapdragon Flight installation

gpanders, 2017-03-10:
Our team is using a Snapdragon Flight. We followed the PX4 installation instructions found on http://dev.px4.io but were having many issues, particularly with the proprietary ESC board; namely, we could not get any communication to the motors through the board. We contacted Intrinsyc for help on this and they told us to follow the installation instructions found on the ATLFlight/PX4 fork (found [here](https://github.com/ATLFlight/ATLFlightDocs/blob/master/PX4.md)). When we followed these instructions (in particular, the instructions for **Building with FC_ADDON drivers**) everything seemed to work as intended.

The instructions on this page are very different than what is listed on the current PX4 dev documentation, and it took us many weeks and having to contact the vendor in order to determine the proper installation procedure.

Why is there a discrepancy between what is on the PX4 dev documentation and what is on the ATLFlight repo? What is the difference between **eagle_legacy_default** and **eagle_default**? Do the instructions on the PX4 dev docs need to be updated so others do not have this problem in the future?

LorenzMeier, 2017-03-13:
@jywilson Are you able to help?

In general we're not supporting the Qualcomm ESC board officially but PWM motor controllers. If Qualcomm wants to change that we would need a couple of test units, which we have not received to date. @mcharleb

jywilson, 2017-03-24:
Thanks for letting us know about this.  

You bring up a very good point.  We need to link to the documentation on github ATLFlight from github PX4.  The documentation on github ATLFlight is more actively maintained by Qualcomm engineers than the docs in the PX4 repo.  I think we need to eliminate the redundancy to avoid this confusion.   

I am going to send a copy of this issue to my coworker so he can also comment.

Jim W.

bharath374, 2017-03-29:
@gpanders Currently there are two ways of building the PX4 code for Snapdragon Flight. The "eagle_legacy_default" builds the PX4 using proprietary drivers for the ESC and RC receiver (this mode is described [here](https://github.com/ATLFlight/ATLFlightDocs/blob/master/PX4.md#building-with-fc_addon-drivers)). The "eagle_default" option builds the code with open-source drivers (this mode is described [here](https://github.com/ATLFlight/ATLFlightDocs/blob/master/PX4.md#building-with-upstream-drivers)). We will aim to add better documentation to make this distinction clear.

gpanders, 2017-03-30:
@bharath374 This is the point I am still confused about: are the open-source drivers used by "eagle_default" meant to be used with [Qualcomm's ESC board](https://shop.intrinsyc.com/products/qualcomm-electronic-speed-control-board)? If not, what is the alternative to the Qualcomm board? To my knowledge, the Snapdragon Flight must be used with this board. Is this not the case?

bharath374, 2017-03-30:
"eagle_default" does NOT support Qualcomm's ESC board. In order to use Qualcomm's ESC board, you need to use "eagle_legacy_default" (i.e. follow the instructions from [here](https://github.com/ATLFlight/ATLFlightDocs/blob/master/PX4.md#building-with-fc_addon-drivers)).

gpanders, 2017-03-31:
@bharath374 Again, what then is the alternative to the Qualcomm ESC board? What do the PX4 developers use if they are not using "eagle_legacy_default"? This is the question I have yet to get a clear answer on. The Snapdragon only has UART output, not PWM, so you cannot use regular ESCs, so what options do you have if you are not using the Qualcomm board?

LorenzMeier, 2017-04-01:
@gpanders Due to the limited availability and initially high price of the Qualcomm ESC board we're using normal PWM ESCs on 250-class racer frames. E.g. this one:
http://www.getfpv.com/multi-rotor-frames/qav250-mini-fpv-quad/qav250-mini-fpv-quadcopter-rtf-carbon-fiber-edition.html

Any 250 racer will do. @ChristophTobler Can you add the 3D printed mounting parts for these frames to the dev guide?

LorenzMeier, 2017-06-15:
I believe we did this quite some time ago @ChristophTobler?

ChristophTobler, 2017-06-16:
@gpanders @LorenzMeier Our build is already in the [dev guide](https://dev.px4.io/en/flight_controller/snapdragon_flight.html) (including the 3D printed mounts)
I'll check with @Stifael about the ESC board and add it to the guide

ChristophTobler, 2017-06-16:
@gpanders can you have a look at the [snapdragon dev guide](https://dev.px4.io/en/flight_controller/snapdragon_flight.html) and tell if there is anything that is not clear?

gpanders, 2017-06-16:
The "Recommended setup" section was added after I posted this issue. That section does clear up a lot of the confusion. 

I still think it would be beneficial to add a note in that section specifying that the ESC board that Qualcomm sells is incompatible with the **eagle_default** build mode. A link to the ATLFlight repo (or at least a mention of it) will help any future teams using the Snapdragon who also purchased the Qualcomm ESC board.

Our team did end up moving away from the Qualcomm board, so this is no longer an issue for us. But having this information listed on the PX4 site would have helped us figure out this problem much sooner.

Thanks guys!

ChristophTobler, 2017-06-19:
@gpanders You are right, I'll add that to the guide. Thanks!

ChristophTobler, 2017-06-19:
added https://github.com/PX4/Devguide/commit/96b7e8d42314cc78f1c2d1ea3222366aa8d46342

closed 2017-06-19 by ChristophTobler