PX4/PX4-Devguide #96: Sidebar links lost

mhkabir, 2017-03-20:
Looks like the timesync and vision pages got lost due to force push or something.

hamishwillee, 2017-04-10:
Hi @mhkabir 

Are these still gone? Can you remember what the name of the file or the heading was? I rebuilt the January version and it is not obvious which pages you might be refering to.

hamishwillee, 2017-06-12:
@mhkabir ? See question above. Or can this be closed?

mhkabir, 2017-06-12:
Whoops missed your comment. The timesync page is still missing. Let me find it.

mhkabir, 2017-06-12:
It doesn't seem to be in the repo anymore; It potentially got lost during the switch to GitBook when the older repo was renamed and eventually deleted.

hamishwillee, 2017-06-12:
@mhkabir  Is it something that could be recovered from here: http://dev.px4.io.s3-website-us-east-1.amazonaws.com/advanced-camera-trigger.html ? If so, which doc(s)

hamishwillee, 2019-02-03:
@mhkabir I'm closing as even if these existed once, I presume they would be too out of date to be of any use now. Also I can't find them :-0

closed 2019-02-03 by hamishwillee