PX4/PX4-Devguide #138: Doc Bug: TBS Caipiroshka · PX4 Developer Guide

hamishwillee, 2017-05-02:
There is a missing zip file "design files" on this page - 

```
* Left and right 3D-printed motor mount ([design files](parts/motor_mounts.zip))
```

Any idea where they might be located?

Bug Page: [TBS Caipiroshka · PX4 Developer Guide](https://dev.px4.io/en/airframes_vtol/caipiroshka.html)

hamishwillee, 2017-05-02:
@bkueng Any ideas?

mhkabir, 2017-05-02:
@tumbili Should have it.

hamishwillee, 2017-05-03:
@tumbili Roman, any idea where we can get those files?

tumbili, 2017-05-03:
@hamishwillee Hi! Let me check on my old laptop, it could be that I have them there. Otherwise @LorenzMeier could still have them.

hamishwillee, 2017-05-08:
Thanks, any luck @tumbili ?

tumbili, 2017-05-08:
@hamishwillee Just sent them to you via Slack.

hamishwillee, 2017-05-08:
Thank you @tumbili . All fixed now

closed 2017-05-08 by hamishwillee