PX4/PX4-Devguide #179: FYI... broken links...

kintzlr, 2017-06-07:
Broken link...
 This page: 
   https://dev.px4.io/en/advanced/realsense_intel_driver.html
 Contains this link:
   RealSense_ROS repository
 That points here:
   https://github.com/PercATI/RealSense_ROS
 and there is no there, there.

A couple lines below that, there is another broken link.
 "Follow instructions given in here".
where the word "here" is a link that's also broken.

Is there an alternate location for these files? Or an equivalent option?

Cheers,
-Randy

hamishwillee, 2017-06-07:
@bkueng I get HTTP404 on that repo so either it is private or deleted. 

There is a repo that is probably the same content here: https://github.com/kaorun55/RealSense-ROS
(based on the fact that it has the install instructions in the same relative "here" location as in our docs (https://github.com/kaorun55/RealSense-ROS/tree/master/r200_install)

That said, there looks like a more official set of docs here: https://github.com/intel-ros/realsense

kintzlr, 2017-06-08:
Thanks for the info! I'll check out those links.

bkueng, 2017-06-08:
I don't know about that repo. I'd probably also go for https://github.com/intel-ros/realsense.