PX4/PX4-Devguide #181: Basic Simulation doc needs clarification

hamishwillee, 2017-06-12:
The doc has some questions that it would be good to answer: https://dev.px4.io/en/simulation/sitl.html

1. This is "Basic Simulation". 
   * In what way is it basic? 
   * What can you do with this simulation and what can't you do that you might expect to be able to
   * What can't you do this with simulation that you can with Gazebo and with ROS + Gazebo (which presumably count as "advanced".

2. Some things I might like to do that I can't see from information provided (please advise if possible or not, and if possible, how):
   * How connect to a GCS
      * Looks like QGC auto detects if on same machine. (is there any way to see "how" QGC is connected?)
      * What UDP is it connecting to? udp:127.0.0.1:14550 ?
      * Are any other GCS known to work - MAVProxy?
      * What if I want to connect GCS from another computer - can I do that?
    * What UDPs are exposed? (e.g. to connect companion computer). Can I add more?
   * How can I change the vehicle simulated? What are the options? (I'm guessing just use QGC firmware, but then the vehicle would need to be calibrated ...)
   * Are there sets of parameters I can preload for different testing? 
      * What options are available there. 
      * How do I load them?
      * Can I use QGC to play with parameters?
   * How can I test various fail states - loss of GPS, crossing geofence,
   * How can I test changes in environment - wind, vibration, magnetic interference, overheating
   * How can I test additional components - rangefinders, gimbals, optical flow sensors, parachutes etc

3. The simulation is demonstrated with `commander takeoff`. 
   * Why is this used for demo - is it just because it is a basic command you can send from the shell?
   * Are there other good commands you can use to move the vehicle around? Or would that all be done via GCS.

4. The command `make posix_sitl_default jmavsim` - what are the "bits" ie "posix_sitl_default" and jmavsim. Are there options for these - e.g. to simulate some other vehicle

5. In (simulating a wifi drone)[https://dev.px4.io/en/simulation/sitl.html#simulating-a-wifi-drone] you can broadcast the IP on the local network. 
   * Is the IP address for companion computer still exposed? 
   * Does this change anything else - ie does this limit or change usage in other ways?

6. In [extending and customising](https://dev.px4.io/en/simulation/sitl.html#extending-and-customizing):
   * It says "To extend or customize the simulation interface, edit the files in the Tools/jMAVSim folder. The code can be accessed through the jMAVSim repository on Github." - are you saying that you should edit the files in the jMAVSim repo or the Tools/jMAVSim folder - it seems both!
   * What extension/customisation might someone reasonably want to do?

7. The "Important Files" sections says the following
   * Why are these important - what can you do with them?
   ```
   The startup script is in the posix-configs/SITL/init folder and named rcS_SIM_AIRFRAME, the default is rcS_jmavsim_iris.
   The root file system (the equivalent of / as seen by the) is located inside the build directory: build_posix_sitl_default/src/firmware/posix/rootfs/
   ```

I got this started in virtualbox with QGC and the latest PX4. However as soon as it takes off I get "Failsafe enabled: no local position" essages. Lots of drifting of my vehicle. EFK Fusion timeouts reported. GPS seems OK in that QGC shows 10 satellites. Is there something I can do to simulation to get everything to work "out of the box" in terms of flying the vehicle?

Stifael, 2017-06-16:
I will try to answer the question above, but I do not know the answer to many of the questions above either.

1.
- basic 
not sure if it is the best name. it just means that the autopilot is simulated via the jmavsim simulator. it probably makes more sense to just call it what it is: jmvasim simulator
- what you can do:
a.) fly around in the simulated world similar to the real world. I think it makes sens to point the user to the website: https://pixhawk.org/dev/hil/jmavsim and repo here:
https://github.com/px4/jMAVSim
b.) camera mount is fixed. this means that the user cannot do a 3d rotation around the vehicle.  but the camera can be placed anywhere (for instance on the vehicle).
c.) supports HIL

2.
udp connection ports:
a.) any time you start px4 sitl (for instance make posix_sitl_default jmavsim) then in most cases three mavlink streams are started on port 14556 (normal mode) and 14557 (onboard mode) (example here: https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/ekf2/iris#L64-L65) and sitl port:
normal mode -> binding port: 14556, remote port: 14550
onboard mode -> binding port: 14557, remote port: 14540
sitl udp port: 14560 (this is a parameter: https://github.com/PX4/Firmware/blob/master/src/modules/simulator/simulator_params.c#L48)

b.) Simulator
udp port 14560 (communicates with px4)

c.) QGC / any GCS
udp port 14550 (communicates over normal mode with px4)

An outdated example is here: https://github.com/PX4/Firmware/tree/master/posix-configs/SITL
would be nice to have something similar.

- connect GCS from another computer:
Yes that is possible as long as both computers support mavlink. You can run QGC on one machine that sends mavlink messages over wifi to an other machine where px4 + sitl runs. 

- add companion computer;
there is a second mavlink stream running by default (see onboard mode above) that can be connected to by any program listening to the port 14540.  An example is given here (launching MAVROS): 
https://dev.px4.io/en/simulation/ros_interface.html
If both mavlink stream are already taken, then I assume the only option is to start a third mavlink stream (@bkueng is that correct?).

- 
> How can I change the vehicle simulated? What are the options? (I'm guessing just use QGC firmware, but then the vehicle would need to be calibrated ...)

The simulated vehicle has to be defined in the simulator (which is a submodule of px4 firmware):
https://github.com/PX4/jMAVSim/tree/361a351360e2a0e1c726b984119e3b057bec01b6/src/me/drton/jmavsim/vehicle

-
> Are there sets of parameters I can preload for different testing? 

do you mean parameters specific for the simulator or for the firmware? firmware specific parameters are specified in the init file for each type of simulation: 
example here: https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/ekf2/iris#L4-L46
This parameters can be changed via QGC as well and can be loaded from QGC in a special file. In QGC, go to parameters, then in the right top corner to Tools -> there is the option "save to file" and "load from file"

- 
> How can I test various fail states - loss of GPS, crossing geofence

certain tests are straight forward, but there is no module that takes care of them (would actually be nice to have in the future). RC loss for instance can be tested by unplugging the controller, GPS loss can be tested by typing in the terminal from which the simulation was started: gps stop
this will stop the gps module from running. 

> How can I test changes in environment - wind, vibration, magnetic interference, overheating

I guess changes like that would first need to be implemented in the simulator. thats rather a question of how good the simulator is (which is a different project).

> How can I test additional components - rangefinders, gimbals, optical flow sensors, parachutes etc

You need to write a new init file that starts the px4 modules required to deal with the additional components:
example here for optical flow: https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/ekf2/iris_opt_flow#L46 where EKF2_HGT_MODE 2 means that optical flow is used for height data.

if you also want to visualize the component, then you need to add a new vehicle model 
 (optical flow example for gazebo here: https://github.com/PX4/sitl_gazebo/tree/02060a86652b736ca7dd945a524a8bf84eaf5a05/models/iris_opt_flow)

I will finish the rest later.

bkueng, 2017-06-19:
> If both mavlink stream are already taken, then I assume the only option is to start a third mavlink stream (@bkueng is that correct?).

Yes I think so. But I'd call it mavlink instance, mavlink stream means something different. It's also worth stressing that the [simulator module](https://github.com/PX4/Firmware/tree/master/src/modules/simulator) is responsible for the mavlink communication with the simulator, for all others (QGC, mavros, ...) it's the `mavlink` module.
We can also do a similar graph for HIL, which uses a similar setup, except over UART and the simulator module is not used and QGC connects to jMAVSim, providing a pass-through channel.

hamishwillee, 2017-06-20:
Thank you @Stifael . That is a really good start.

> You need to write a new init file that starts the px4 modules required to deal with the additional components:

8. You point to an init file that does optical flow already ([/SITL/init/ekf2/iris_opt_flow](https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/ekf2/iris_opt_flow#L46)). So 👍 
   - presumably I don't need to write a new init file but can use that one, or set that parameter via QGC?
   - how do I specify which init file(s) to use on command line?
   - Is there any information about what these various init files are for? Ie so I can say "if you want to specify an Optical thingy use X, if you want to specify something else, use Z"
   - The docs say 
     > The startup script is in the posix-configs/SITL/init folder and named rcS_SIM_AIRFRAME, the default is rcS_jmavsim_iris.

     That doesn't exist. How do I work out what is started by default?

9. It is not completely clear what is provided by SITL and what is provided by jmavsim or gazebo. What I think (and hope you can confirm) is that:
   -  SITL is a build of PX4 running on your computer. 
      - It sets up a bunch of mavlink connections, "normal" which is to talk to a GCS, offboard which is to talk to a companion computer, sitl, which is to talk to the simulation engine (jmavsim, gazebo). 
      - What does remote port vs binding port mean? binding port: 14556, remote port: 14550
      - You can set up additional links in the init file. 
      - The init file can set up various services to run (e.g. like the optical sensor or a gimbal or whatever)
         - can you connect an RC control to SITL?
      - All the inputs to the SITL hosted version of PX4 come from the simulator (jmavsim, gazebo, whatever). 
         - So the SITL PX4 will get GPS, compass, optical flow info, whatever as mav messages from the Simulator.
         - The SITL PX4 will sent simulated AUX, MAIN outputs via the sitl mavlink to the simulator, which can then calculate how thrust etc.
         - The SITL PX4 will send the normal mavlink messages to offboard and QGC, based on its simulated inputs. 
   - Jmavsim is a simulator running on a computer/network
      - It is connected to the vehicle over a network (? or via serial/USB?).
      - It sends position, direction, vehicle orientation information simulated from expected movement of vehicle based on outputs from MAIN/AUX.
      - What else does it send?
      - Will it simulate an image being taken/camera triggering?
      - For optical flow, does it simulate an optical flow, or just send a speed and PX4 interprets that as coming from an optical flow sensor IFF you have that module enabled?
   - HITL essentially replaces the comms link from SITL to a real flight controller
      - The flight controller generates outputs from simulated inputs from Jmavsim. Everything else is the same (ie the only difference is that the outputs reflect real hardware responsiveness to the inputs.

1. Re "what can you do with it"

>  fly around in the simulated world similar to the real world.

Well yes, but presumably flying around is "for a purpose" - which is to test something. So perhaps the question would be better as "what can you test and what can't you test"?

From above it sounds like jmavsim provides ability to simulate gps, compass, gyro. It sounds like you can turn off modules to fake failure of components. It sounds like you can simulate lidar (?) but you can't actually set up buildings or anything to avoid. Wind etc is not part of the physics engine.

Essentially, if the vehicle does fly in this sim then you know that the estimator and most of your other settings are OK.

What else have you used it to verify?

Sorry to be so long winded. We can chat over this in a call if it suits you better.