PX4/PX4-Devguide #182: Gazebo doc needs clarification

hamishwillee, 2017-06-12:
Doc is: https://dev.px4.io/en/simulation/gazebo.html

1. Docs say `supports standalone use (without ROS) or SITL + ROS`. Would it be more correct to say `Gazebo supports integration to SITL either with or without ROS`.
2. What can you do with Gazebo for testing that you can't do with the basic simulator?
   - Why is it better?
   - What vehicles can you simulate. 
   - Is changing/simulating a vehicle configured in SITL or Gazebo or both? How? I think something to do with the start commands `make posix_sitl_default gazebo_tailsitter` but don't know what arguments are/options.
   - Can you simulate all the things I mentioned in #181 - wind, compass failure, GPS failure etc - and if so, in what way/where?
3. As I understand it ROS adds ability to use offline mode and drive the vehicle remotely using scripts/components. So I guess the main reasons for us are to add automation both for "super flight modes" (as per dronekit) and for automated testing. 
   * Is that correct?
   * Anything else that makes it useful?
   * If you install ROS (as per the "interfacing" doc can you still invoke Gazebo "as normal" in this doc
4. When you have Gazebo started, do you fly it in just the same way as for the Basic simulation - e.g. via GCS? What I'm heading towards here is that it might be worth separating out the doc on "how to fly the sim" for use by both. 
5. Gazebo 8 is coming along. Do we have a plan to support it.
6. Gazebo 7 is recommended everywhere. Would we be better off presenting the Gazebo 7 option and having Gazebo 6 as an "aside" - or to put it otherwise, why do we ever suggest Gazebo 6?
7. In the section [Change World](https://dev.px4.io/en/simulation/gazebo.html#change-world) it says "The current default world is the iris.wold located in the directory worlds. The default surrounding in the iris.world uses a heightmap as ground. This ground can cause difficulty when using a distance sensor. If there are unexpected results with that heightmap, it is recommended to change the model in iris.model from uneven_ground to asphalt_plane."
   * How do you change the iris model?
   * How do you change the default world?
   * The video shown here has an interesting world, not just an empty plane. How do I get a world like that?
8. In extending and customising section it says: "To extend or customize the simulation interface, edit the files in the Tools/sitl_gazebo folder. The code is available on the sitl_gazebo repository on Github."
   * What customisation might you want to do
   * Are there any good docs on customisation
   * Do you edit in the repo or in the subfolder? I guess you do basic tests in the subfolder but you have to commit to the repo to get changes into the master?

ChristophTobler, 2017-06-12:
> What can you do with Gazebo for testing that you can't do with the basic simulator?

You can simulate sensors such as lidar, sonar, cameras etc. -> e.g. `make posix_sitl_default gazebo_iris_opt_flow`

hamishwillee, 2017-06-12:
@ChristophTobler That is cool. Is there any docs anywhere on the "scope" of what the sensor simulation can do. So for example I would want the lidar/sonar to provide me realistic distance estimation around my world. I might want to be able to simulate it to the level of realism that it is inaccurate out past a certain distance. Similarly a basic camera simulation might just echo back that images have been taken, while a really good one might actually take a video of the simulated environment and feed it to right locations ...

ChristophTobler, 2017-06-12:
There is the Gazebo sensors documentation http://osrf-distributions.s3.amazonaws.com/gazebo/api/dev/classgazebo_1_1sensors_1_1Sensor.html
You can add noise to the sensors (we do). The simulated camera provides an actually simulated image which then can be used for computer vision tasks as e.g. flow, VIO, avoidance etc.

hamishwillee, 2017-06-12:
Thanks @ChristophTobler - presumably you can't do as much in the basic sim to define the environment either - e.g. maybe wind etc, but not building etc?

ChristophTobler, 2017-06-12:
I've never seen someone simulating wind in Gazebo but buildings etc. is super easy. You can just drag and drop them from a database http://models.gazebosim.org/ (You can also see them in our simulations e.g. https://www.youtube.com/watch?v=tGF24r1PTRs)

hamishwillee, 2017-06-12:
Thanks @ChristophTobler  - are there any of the other questions above that you can help answer?

andre-nguyen, 2017-06-13:
I'll take a stab at it:

>     Docs say supports standalone use (without ROS) or SITL + ROS. Would it be more correct to say Gazebo supports integration to SITL either with or without ROS.
Yes

>     What can you do with Gazebo for testing that you can't do with the basic simulator?
Simulate many kinds of sensors and complex environments.

>         Why is it better?
Because then you can do complex behaviors other than simple waypoint navigation. E.g. obstacle avoidance or visual state estimation.

>         What vehicles can you simulate.
Not entirely sure but a good place to start is here https://github.com/PX4/sitl_gazebo/tree/master/worlds so at the very least:

* Delta wing
* iris
* iris with optical flow
* iris with rplidar
* rover
* 3dr solo
* standard vtol (quadplane)
* tailsitter
* typhoon h480

>         Is changing/simulating a vehicle configured in SITL or Gazebo or both? How? I think something to do with the start commands make posix_sitl_default gazebo_tailsitter but don't know what arguments are/options.
Basically when you make the target, at some point this [script](https://github.com/PX4/Firmware/blob/ca05e64a9f1464f134b1500a03ee42c718c7a2e9/Tools/sitl_run.sh#L94) is run and it calls a gazebo world containing the appropriate vehicle. So the make target itself is the "option" for running gazebo with the vehicle you want.

>         Can you simulate all the things I mentioned in #181 - wind, compass failure, GPS failure etc - and if so, in what way/where?
There is a [wind plugin](https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_wind_plugin.cpp) but I don't think it is readily accessible through a make target. But it would be a simple case of making a world file that includes that plugin. For the other things, they are possible in theory but I don't think its implemented. 

>     As I understand it ROS adds ability to use offline mode and drive the vehicle remotely using scripts/components. So I guess the main reasons for us are to add automation both for "super flight modes" (as per dronekit) and for automated testing.
I'm not sure what offline mode is? I am not familiar with super flight mode. But as mentioned further up, you can write your own program to control the vehicle in the way you want. Also some gazebo sensor plugins support publishing their data directly to ros. So a ros process could listen in on a gazebo sensor (e.g. a lidar) and command the vehicle (through mavros) to do obstacle avoidance.

>         Is that correct?
>         Anything else that makes it useful?
I mean, it looks cool too.

>         If you install ROS (as per the "interfacing" doc can you still invoke Gazebo "as normal" in this doc
>     When you have Gazebo started, do you fly it in just the same way as for the Basic simulation - e.g. via GCS? What I'm heading towards here is that it might be worth separating out the doc on "how to fly the sim" for use by both.
>     Gazebo 8 is coming along. Do we have a plan to support it.
(Personal opinion) it's probably best to just follow the Gazebo version paired with the latest ros version. So apparently [both ros Lunar and Kinetic go with Gazebo 7](http://wiki.ros.org/Distributions).

>     Gazebo 7 is recommended everywhere. Would we be better off presenting the Gazebo 7 option and having Gazebo 6 as an "aside" - or to put it otherwise, why do we ever suggest Gazebo 6?
Probably a historical artifact from when the docs were written. At the time I think ros was at version indigo/jade and the gazebo versions were 2.x/5.x . Some plugins for sensor simulation and physics simulation needed the latest Gazebo to work.

> In the section Change World it says "The current default world is the iris.wold located in the directory worlds. The default surrounding in the iris.world uses a heightmap as ground. This ground can [...]
Off the top of my head you change the make target. e.g. `make posix_sitl_default gazebo_iris` to `make posix_sitl_default gazebo_delta_wing`

>         How do you change the default world?
Modify or add a world [here](https://github.com/PX4/sitl_gazebo/tree/master/worlds)

>         The video shown here has an interesting world, not just an empty plane. How do I get a world like that?
You can build it in the Gazebo editor.

>     In extending and customising section it says: "To extend or customize the simulation interface, edit the files in the Tools/sitl_gazebo folder. The code is available on the sitl_gazebo repository on Github."
>         What customisation might you want to do
Everything you said previously, e.g. add obstacles in the world, change the heightmap and textures, change the vehicles with more or less sensors...

>         Are there any good docs on customisation
Nope lol. But I guess the official Gazebo tutorials are a good place to start.

>         Do you edit in the repo or in the subfolder? I guess you do basic tests in the subfolder but you have to commit to the repo to get changes into the master?
Not sure. I never really knew what to do if you had a fork of a project and you wanted to also fork the submodule, but you also wanted to keep up with the upstream submodule.

hamishwillee, 2017-06-19:
Thanks @andre-nguyen ! I missed this response. I will be integrating fixes over coming week.

dan62, 2017-10-28:
hey guys im trying to get the bottom arial camera to display from the optical flow simulation in gazebo like this link [(https://www.youtube.com/watch?v=tGF24r1PTRs)] however my camera is not displaying the ground captured image, please help i am using PX4 ROS for quadrocopters