PX4/PX4-Devguide #191: External Position Estimation needs to be updated

eric1221bday, 2017-06-18:
The current documentation in the DevGuide in regards to External Position Estimation is quite outdated. In particular since EKF2 is now feature complete and the default estimation engine, the guide should at least be extended to cover external position estimation under EKF2 in addition to the existing docs for LPE.

mhkabir, 2017-06-19:
We will update it. LPE is still the recommended estimator for motion capture and vision use-cases till we have more testing on EKF2. e.g https://github.com/PX4/Firmware/issues/7408

bramsvs, 2018-06-21:
@mhkabir Maybe time to update this? I'm happy to help but I have to get it working first ;)

hamishwillee, 2018-06-22:
... and probably LPE no longer the recommended estimator @mhkabir

mhkabir, 2018-06-22:
Yeah, we have EKF2 dialed in now and should be the recommend estimator.

hamishwillee, 2018-06-22:
Thanks @mhkabir . Can you help @bramsvs get started?

m0oz, 2018-10-31:
Hey there. Just chiming in.

It seems like this is still relevant. I haven't contributed to the doc yet but I am about to start my implementation of external pose estimate + EKF2 and I could add my findings to
https://dev.px4.io/en/ros/external_position_estimation.html

**Regarding the overall structure of the doc on external vision:**
To me it was counter-intuitive to find the doc on external vision split between 
[Robotics/ROS/ExternalPositionEstimate](https://dev.px4.io/en/ros/external_position_estimation.html) and [Tutorials/eclEKF](https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html)
I think https://dev.px4.io/en/tutorials would need an additional page that gives more details on how to setup your vio->mavros->px4 pipeline and how to tune ekf2 for this application.
Robotics/ROS/ExternalPositionEstimate could then just link to this tutorial as it currently reads as a tutorial anyways.

I think this supports what was mentioned here #530 @hamishwillee

hamishwillee, 2018-11-01:
Hi @m0oz 

We definitely need clear docs on external vision. It is completely appropriate for a separate EKF tutorial, but the important aspects should either be included or strongly linked from the external position docs.
I am not sure having a separate document for vio->mavros->px4 makes sense - it probably can be merged - or the common bits should be split.

I should also add that I am trying to kill that whole section on tutorials by moving the docs into sections with associated content and making them more guide than tutorial. I don't hate tutorials, but there is a tendency for them to cover just one setup case and not have enough information for people to extend them to more general use cases. 

With respect to the ExternalPositionDoc, is any of what you're proposing covered in the updated in https://github.com/PX4/Devguide/pull/573 ?

Cheers
H 

PS I appreciate my response is a bit superficial. Once you've looked at #573 I will think about this more "completely". Your help/experience on getting the information for this updated will be invaluable.

PR #681 (2018-12-19): [WIP] Vision/MoCap docs improvements
This is update to Vision/Mocap pages following on from #573.

Intention is to more concisely separate into what is supported by PX4 and the estimators, and then map this to ROS. This might be done in separate docs, one under computer vision and one under ROS [TBD]

Next step for iterating this is when there are answers to:
* [x] https://github.com/PX4/Devguide/pull/573#issuecomment-443933299
* [x] https://github.com/PX4/Devguide/pull/573#discussion_r238493427
* [x] Capture "For EKF2, the only frame_id supported is the one that matches MAV_FRAME_VISION_NED,  [... cut]
closed 2018-12-19 by hamishwillee