PX4/PX4-Devguide #223: Tuning the ECL EKF mentions unexisting parameters

nicolaerosia, 2017-07-18:
Page [here](https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html)
The `Drag Specific Forces` chapter mentions EKF2_AID_MASK bit 4, but that is used for "4: vision yaw fusion", https://dev.px4.io/en/advanced/parameter_reference.html
EKF2_BCOEF_X and EKF2_BCOEF_Y do not exist.

nicolaerosia, 2017-10-11:
@hamishwillee FYI

hamishwillee, 2017-10-11:
Thanks @nicolaerosia - lets fix this in #319 (I've asked @priseborough for help).

priseborough, 2017-10-13:
Bit 5 of EKF2_AID_MASK is used to enable the multi-rotor drag fusion. See: https://github.com/PX4/ecl/blob/160e4d69c1bab6daebedbff2de3c4e08c438eefa/EKF/common.h#L176

EKF2_BCOEF_X and EKF2_BCOEF_Y exist here: https://github.com/PX4/Firmware/blob/master/src/modules/ekf2/ekf2_params.c#L1082
https://github.com/PX4/Firmware/blob/master/src/modules/ekf2/ekf2_params.c#L1094

nicolaerosia, 2017-10-13:
Thanks, I was only looking into the docs which seem to be outdated