PX4/PX4-Devguide #225: Tutorial for debugging with Black Magic Probe and Qt Creator

master-of-desaster-88, 2017-07-20:
For beginners in debugging, like me, it would be very helpfull to have a manual how to connect the probe on a PX4 (for example Pixhwak) and start the first debugging with Qt Creator on a Linux System (Ubuntu). 
I found a few hints on the "pixhawk.org" site and as a electric engineer I also like the "try and error" system. But sometimes you get a moment where nothing will work and then it get's really frustrating.
So it would be nice if I could found something at the very nice PX4 Developer guide.

Thanks and best regards,
Markus

burkea, 2019-04-02:
Dear Markus,
I have the same problem. Could you give me brief instructions, please?
Best.
Burke

master-of-desaster-88, 2019-04-02:
Dear Burke,
So far I didn`t get a answer to my request.
Still same level.
BR, Markus

hamishwillee, 2019-04-04:
FYI, I started a discussion on this on slack, in #general channel: https://px4.slack.com/archives/C0V533X4N/p1554262930087000

The general consensus seems to be that you're better off using the Dronecode probe than Black Magic Probe. This is because it has the same connector standard as pixhawk - so you just plug it in. There is also some comment that BMP is not great with recent pixhawks because the probe is not well supported. 

I don't know about IDE integration at all. Some info about using the Dronecode probe is here: https://dev.px4.io/master/en/debug/system_console.html#connecting-via-dronecode-probe

smooker, 2020-09-30:
https://github.com/smooker/bmpqtcreator
Ugly, but works.

hamishwillee, 2020-10-01:
The main IDE PX4 is now recommending is VS Code, which is documented here: https://dev.px4.io/master/en/setup/vscode.html. I am trying to get info on how the hardware debugging is set up in https://github.com/PX4/Devguide/pull/1092

FYI, Info on probe setup is now documented here https://dev.px4.io/master/en/debug/swd_debug.html - so any docs about IDEs can cross link to that.

@smooker Thanks for that - however most users will need more than that to know how to get started. I.e. how you would install this patch, when, where, what versions of QT creator it would live on etc. We'd welcome a full Qt creator ID page like this one: https://dev.px4.io/master/en/setup/vscode.html if you ever have time.

hamishwillee, 2020-11-04:
I am closing this as VSCode is now our recommended dev platform. If someone wants to add this tutorial that would be welcome, but not going to leave this as an action.

closed 2020-11-04 by hamishwillee