PX4/PX4-Devguide #232: PX4 Docker containers

Stifael, 2017-07-25:
I am not sure what is missing when i try to set up a docker container with mapping my local_src directory to the container. 
The steps that I took are pretty much the same as in the documentation [here](https://dev.px4.io/en/test_and_ci/docker.html).

1.) pull docker file:
`docker pull px4io/px4-dev-ros:2017-05-31`

2.) 
`xhost +`
`docker run -it --privileged \`
`-v <src_to_local_firmware>:/src/firmware/:rw \`
`-v /tmp/.X11-unix:/tmp/.X11-unix:ro \ `
`-e DISPLAY=:0 \`
`-p 14556:14556/udp \`
`--name=px4_ros_sim px4io/px4-dev-ros:2017-05-31 bash`

I also tried to pull different images of the px4-dev-ros (the latest and v1.0). In all attempts I successful entered the container and the Firmware was in the `/src/firmware` as expected, but I was not able to build anything but received the error that no Makefile was found (despite having a Makefile in the directory)

hamishwillee, 2017-07-26:
G'day, I just followed the instructions on this too (my they assume you know a lot about docker) - running inside a Ubuntu 16.04 VM on Windows. 

When you say that the makefile directory was present - you mean that you can see it within docker?

I entered near-identical instructions - the main difference being use of sudo and also specified the source tree:
```
xhost +
sudo docker run -it --privileged \
-v ~/src/Firmware:/src/firmware/:rw \
-v /tmp/.X11-unix:/tmp/.X11-unix:ro \
-e DISPLAY=:0 \
-p 14556:14556/udp \
--name=px4_ros_sim px4io/px4-dev-ros:2017-05-31 bash
```
When I try  build anything I don't get missing Makefile errors ...
```
make posix_sitl_default gazebo
```
Instead I get permission denied errors when make tries to create the build directories (followed by recipe fails). Any ideas on that?

hamishwillee, 2017-07-26:
PS Just some notes on docs for when/if I come to document this:
1. "Pull one of the tagged images if you're after a container that just works" - how do I get a list of tagged images? Link to an example list
2. The list of containers in our docs does not match github - ie hierarchy [here](https://github.com/PX4/containers/tree/master/docker/px4-dev#container-hierarchy)
   * In particular, our docs refer to px4io/px4-dev , but heirarchy does not list this ... e.g. it has px4-dev-nuttx
3. A real example using the normal ~/src/Firmware/ cloning/installation would not hurt. Even a bash script to do everything/set this up. 
4. For Linux the docker instructions say that you have to use `sudo` for all instructions from host. Need to make that clear.

Stifael, 2017-07-26:
@hamishwillee 
> When you say that the makefile directory was present - you mean that you can see it within docker?

Yes. If I am within the docker container px4_ros_sim and go to the directory `/src/firmware` then I basically can see the normal px4 Firmware directory (including the Makefiel). But I cannot build anything...

Not sure why you get permission denied or even why you should be root.

hamishwillee, 2017-07-26:
> Not sure why you get permission denied or even why you should be root.

From [docker install guide](https://docs.docker.com/engine/installation/linux/docker-ce/ubuntu/#install-using-the-repository): "You need to use sudo to run Docker commands. Continue to Linux postinstall to allow non-privileged users to run Docker commands and for other optional configuration steps."

Stifael, 2017-07-26:
but `make` is not a docker command

hamishwillee, 2017-07-26:
Sorry, I was unclear. I ran docker itself using sudo (i.e. `sudo docker run -it ...` - your example doesn't. I think may try to add a group as per docs and see if that changes effect of permissions

Stifael, 2017-07-26:
true: this is needed
https://docs.docker.com/engine/installation/linux/linux-postinstall/

hamishwillee, 2017-07-26:
Actually it wasn't enough - I still got permission errors. I've just used chmod to give full access to my current user over the host folder (not clear why they didn't have it). Everything building now.

hamishwillee, 2017-07-26:
Sorry, didn't mean to close!

Stifael, 2017-07-26:
@hamishwillee 
with `sudo docker run ...` I also get the permission denied error

hamishwillee, 2017-07-31:
@Stifael Don't know if you saw, but Daniel's scripts for CI might be helpful: https://github.com/PX4/containers/issues/65#issuecomment-318457659