PX4/PX4-Devguide #239: Add pyulog to ubuntu_sim_nuttx.sh

CarlOlsson, 2017-08-03:
Would be nice if [pyulog](https://github.com/PX4/pyulog) was included in [ubuntu_sim_nuttx.sh](https://dev.px4.io/en/setup/dev_env_linux.html)

hamishwillee, 2017-08-03:
Hi Carl, 

I've tried to keep the stuff in those scripts near-minimal for the developer environment. My concern is that they grow out of of control with whatever people think might be useful for some developer. 

That said, 
1. I'm very happy with this if it is explained in the [manual setup](https://github.com/PX4/Devguide/blob/master/en/setup/dev_env_linux_ubuntu.md) or possibly as part of "additional tools". Why do you think this should be part of the dev env?
2. Is this useful only in this script, or would it also be useful to simulation-only users, windows users, snapdragon users etc? If so, we also need to update the other scripts: https://github.com/PX4/Devguide/tree/master/build_scripts.

Make sense?

CarlOlsson, 2017-08-04:
Hi,

Okey yea if the philosophy is to keep the scripts near-minimal then it should not be there, feel free to close here. I proposed to add it since I think the majority of developers will need it. It makes it much easier to quickly analyse logs etc.

Yes if we decide to add it it should for sure be added to all scripts.

hamishwillee, 2017-08-07:
@bkueng What do you think - will the majority of developers want to install pyulog into their development environment?

bkueng, 2017-08-07:
> @bkueng What do you think - will the majority of developers want to install pyulog into their development environment?

We can add it, yes - it's a small package and only has few dependencies. It's especially useful if you want to have a closer look at logs, or convert them to CSV.

hamishwillee, 2017-08-08:
Awesome. OK @CarlOlsson - I have added this as https://github.com/PX4/Devguide/pull/249 (now merged).  If you could give the changes a quick scan, that would be great.

closed 2017-08-08 by hamishwillee