PX4/PX4-Devguide #268: Fixes to SBus Linux Driver doc

hamishwillee, 2017-08-22:
Updated circuit diagram - with labels showing on what device we're connecting
https://github.com/PX4/Devguide/pull/263#issuecomment-323653386

@crossa  Make Chinese version match English version
https://github.com/PX4/Devguide/pull/263#issuecomment-323719864

Create PCB that replaces the breadboard 
@crossa @TSC21

TSC21, 2017-08-22:
Thanks Hamish. As soon as I have some bandwidth I can create the PCB schematics and layout. Where can those be upload then?

hamishwillee, 2017-08-22:
They can go in the devguide "somewhere". Depending on file format/size maybe zip an put in the assets/ directory. 
BUT, if you're linking to anything that isn't an image then you'd link to the raw file on github (rather than as a relative directory link) because gitbook doesn't download files nicely.