PX4/PX4-Devguide #286: Add a "MAVLink" section in "Concepts"

JonasVautherin, 2017-09-07:
In #285, I mention that a section about MAVLink would be helpful somewhere. On Slack, @hamishwillee  mentioned creating a "Setting up MAVLink Ports & Streams" section in the "Simulation" topic.

In my opinion, that could be a new section in "Concepts", that would clarify the following points:
* How exactly does the MAVLink connection work?
    * Here we could have a simplified image (maybe the same as [here](https://dev.px4.io/en/simulation/#sitl-simulation-environment)?) showing that the PX4 on SITL broadcasts on 14550 (meant for QGC), but also that the SITL startup file (as shown [here](https://dev.px4.io/en/simulation/#example-startup-file) can specify other MAVLink interfaces.
    * We would also explain that usually (AFAIK, at least), we use (src: 14556, dst: 14550) for QGC and (src: 14557, dst: 14540) for others like Mavros or DroneCore.
    * I'd like to know here if the startup file is only for the SITL or also for the PX4 itself. In other words, does my Pixhawk use (src: 14556, dst: 14550) and (src: 14557, dst: 14540) by default, or not?  I don't know how this actually works.
    * We would explain here that the client usually waits for a broadcast on 14550 or 14540, but it can also connect directly if it knows the port on the PX4 side (which is usually 14556 or 14557 in the SITL config files).
* How to configure a MAVLink interface?
    * Here, assuming that the PX4 also has a startup script, we would explain in details the kind of commands we can do. Or at least point to a detailed reference.
* MAVLink protocol and alternatives
    * I would see something explaining what MAVLink provides or not, maybe links to useful resources, and a comparison with using FastRTPS (e.g. "MAVLink is more stable, but FastRTPS is more flexible").

JonasVautherin, 2017-09-07:
I just realised there is a [MAVLink Messaging](https://dev.px4.io/en/middleware/mavlink.html) section in "Middleware and Architecture". This one is very low level as it explains how to create mavlink messages from code. I think it still makes sense to have a higher level introduction in "Concepts".

JonasVautherin, 2017-09-07:
And [Communication](https://dev.px4.io/en/middleware/modules_communication.html#mavlink) in "Modules & Commands" has a section for mavlink :-).

hamishwillee, 2017-09-10:
Yes, MAVLink does apply to both SITL and hardware so should (primarily) be documented outside of SITL. The SITL docs should provide brief overview, SITL specific information, and link to generic doc. 

1. The main mavlink doc is the one in Middleware: https://dev.px4.io/en/middleware/mavlink.html
- this explains how but not where to set up the streaming rates.
- this does not explain how or where to set the ports
- This should cross-link to SITL.
- This should cross-link to https://dev.px4.io/en/middleware/modules_communication.html#mavlink
- This could have image like the PX4 one showing either ends of the commands.
- This have an overview covering main points of UDP and how it works on PX4. 
2. The [module doc](https://dev.px4.io/en/middleware/modules_communication.html#mavlink) should cross link to the middleware doc 
3. [Middleware (Concepts)](https://dev.px4.io/en/concept/middleware.html) should be extended to explain a bit more about middleware and link to the "Middleware and Architecture" section (that currently has no doc, so one should be added, in order to create a link).

JonasVautherin, 2017-09-12:
1. I agree with most of the bullet points. If it is the main mavlink doc, I believe that it should explain high-level stuff like the ports for SITL etc. Not necessarily in great details (it would cross-reference the SITL page and the module doc), but still. It should also explain what to expect with the hardware (as opposed to the simulation). Things such as "Is it UDP, TCP, serial?", or "Where is it initialized/defined?". As far as I understand, it is a bit different for each hardware. But it give pointers to e.g. the code that it responsible for that in the PX4. How bad is it to "duplicate" the high-level information and images between those sections?
2. Agreed.
3. What about a high-level explanation of MAVLink for Simulation, MAVLink on hardware (cross-referencing the main mavlink doc), and also then FastRTPS and uORB?

hamishwillee, 2017-09-13:
With respect to 1

> I agree with most of the bullet points. If it is the main mavlink doc, I believe that it should explain high-level stuff like the ports for SITL etc. Not necessarily in great details (it would cross-reference the SITL page and the module doc), but still. It should also explain what to expect with the hardware (as opposed to the simulation). Things such as "Is it UDP, TCP, serial?", or "Where is it initialized/defined?". As far as I understand, it is a bit different for each hardware. But it could give pointers to e.g. the code that it responsible for that in the PX4. 

I don't know what is the right breakdown of information between docs. However I'd say that we write the information we'd find useful, and then find a home :-)

The doc currently is almost entirely absorbed with creating custom messages and provides no context for other usage/uses of MAVLink. So almost anything we add will be useful :-) 

> How bad is it to "duplicate" the high-level information and images between those sections?

It is good not to duplicate too much. As above though, you write the info required, then work out if duplicating is useful.

With respect to 3 

>> Middleware (Concepts) should be extended to explain a bit more about middleware and link to the "Middleware and Architecture" section (that currently has no doc, so one should be added, in order to create a link).

> What about a high-level explanation of MAVLink for Simulation, MAVLink on hardware (cross-referencing the main mavlink doc), and also then FastRTPS and uORB?

I don' think we'd break out MAVLink topics here into Sim vs Hardware - that would be for lower level topics. We would talk about the other types of middleware like uORB etc.

Next steps?