PX4/PX4-Devguide #305: px4_simple_app not listed in Builtin Apps

KochC, 2017-10-08:
Hey, 
I try to create an own application for my DroPix, so I started with the "Hello Sky" Tutorial. First my compiler whats a CMakeList.txt file which is not described in the tutorial, but I was able to create one on my own 😆

px4_simple_app.c
```
#include <px4_config.h>
#include <px4_tasks.h>
#include <px4_posix.h>
#include <unistd.h>
#include <stdio.h>
#include <poll.h>
#include <string.h>

#include <uORB/uORB.h>
#include <uORB/topics/sensor_combined.h>
#include <uORB/topics/vehicle_attitude.h>

__EXPORT int px4_simple_app_main(int argc, char *argv[]);

int px4_simple_app_main(int argc, char *argv[])
{
    PX4_INFO("Hello Sky!");
		
    return OK;
}
```

CMakeLists.txt
```
px4_add_module(
	MODULE examples__px4_simple_app
	MAIN px4_simple_app
	STACK_MAIN 2000
	SRCS
		px4_simple_app.c
	DEPENDS
		platforms__common
	)
```

nuttx_px4fmu-v2_default.cmake
```
# Tutorial code from
# https://px4.io/dev/px4_simple_app
examples/px4_simple_app
```

Compiler prints correctly compiled:
```
make px4fmu-v2_default
[5/5] Creating nuttx_px4fmu-v2_default.px4
```

and the upload seems to work:

```
Firmware chris$ make px4fmu-v2_default upload
[0/1] uploading /Users/chris/Desktop/Wingcopter/Software/PX4/Firmware/build/px4fmu-v2_default/nuttx_px4fmu-v2_default.px4
Loaded firmware for 9,0, size: 1031060 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Attempting reboot on /dev/tty.usbmodem1 with baudrate=57600...
If the board does not respond, unplug and re-plug the USB connector.
Found board 9,0 bootloader rev 4 on /dev/tty.usbmodem1
ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff type: ÿÿÿÿ
idtype: =FF
vid: ffffffff
pid: ffffffff
coa: //////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////8=

sn: 002d00323334510b32363238

Erase  : [====================] 100.0%
Program: [====================] 100.0%
Verify : [====================] 100.0%
Rebooting.
```

but I do not have the px4_simple_app in my Buildin Apps now:
```
nsh> help
help usage:  help [-v] [<cmd>]

  [           date        free        mount       rmdir       true        
  ?           df          help        mv          set         umount      
  break       echo        kill        mw          sh          unset       
  cat         exec        ls          ps          sleep       usleep      
  cd          exit        mkdir       pwd         test        
  cp          false       mkfatfs     rm          time        

Builtin Apps:
  adc
  bl_update
  bst
  camera_feedback
  camera_trigger
  commander
  config
  cu
  dataman
  ekf2
  ets_airspeed
  fmu
  frsky_telemetry
  fw_att_control
  fw_pos_control_l1
  gnd_att_control
  gnd_pos_control
  gpio_led
  gps
  hardfault_log
  hmc5883
  l3gd20
  land_detector
  lis3mdl
  ll40ls
  load_mon
  logger
  lsm303d
  mavlink
  mc_att_control
  mc_pos_control
  mixer
  mpu6000
  mpu9250
  ms4525_airspeed
  ms5525_airspeed
  ms5611
  mtd
  navigator
  nshterm
  param
  perf
  pwm_input
  pwm
  pwm_out_sim
  px4flow
  px4io
  reboot
  rgbled
  sdp3x_airspeed
  send_event
  sensors
  sercon
  serdis
  sf0x
  teraranger
  tone_alarm
  top
  ulanding_radar
  uorb
  ver
  vmount
  vtol_att_control
nsh> 
```

Where is my fault? 😩🤔

closed 2017-10-08 by KochC

KochC, 2017-10-08:
moved to [PX4/Firmware](https://github.com/PX4/Firmware/issues/8087)

hamishwillee, 2017-10-09:
@KochC Thanks for this. The missing make file is a bug - posted in #308 . If you run into other doc issues with this page can you please add them there?

Also, as an aside, feel free to link to the problem page. If you click the "bug" icon on the top of the problem page in docs you'll get an auto-seeded issue with that link.