PX4/PX4-Devguide #308: Hello Sky Tutorial - docs improvements

hamishwillee, 2017-10-09:
Bug Page: [Writing an Application · PX4 Developer Guide](https://dev.px4.io/en/tutorials/tutorial_hello_sky.html)

As discussed in https://github.com/PX4/Devguide/issues/305 the docs are missing description of CMakeLists.txt and should also include link to [final code](https://github.com/PX4/Firmware/tree/master/src/examples/px4_simple_app) for comparison.

anksankita, 2019-03-20:
My problem is px4_simple_app not listed in Built in apps....please give me the solution

hamishwillee, 2019-03-21:
This issue is closed, because the original error (missing make file) was fixed well over a year ago. 

@anksankita The guide and was updated well after this error so it is no longer valid. I am therefore closing it. If the app is missing it is most likely because it is not included in the firmware as described in https://dev.px4.io/master/en/apps/hello_sky.html#build-the-applicationfirmware
I would guess that you're targetting FMUv2 where it is not present by default, and you will need to enable it in your custom build.

closed 2019-03-21 by hamishwillee