PX4/PX4-Devguide #311: Download ubuntu_sim_nuttx.sh missing

DanSartori, 2017-10-10:
In the Toolchain Installation/Linux (https://dev.px4.io/en/setup/dev_env_linux.html), section Pixhawk/NuttX and jMAVSim/Gazebo Simulation, the link to the file ubuntu_sim_nuttx.sh is not working.
It was working 10 days ago

Thanks

DanSartori, 2017-10-10:
I guess now the file is here

https://github.com/PX4/Devguide/tree/master/build_scripts

hamishwillee, 2017-10-10:
Thanks very much! This was fixed by https://github.com/PX4/Devguide/pull/312 (the file hasn't moved, but the documentation was updated incorrectly in a "bulk fix" for an unrelated issue)

hamishwillee, 2017-10-10:
(PS it will take a few minutes for the docs to rebuild)

closed 2017-10-10 by hamishwillee

DanSartori, 2017-10-11:
Thank you for fixing it!

hamishwillee, 2017-10-11:
I'd love to claim credit, but we have @ShikherVerma to thank!