PX4/PX4-Devguide #340: Can't change to OFFBOARD Mode in Ros Gazebo

billiepratama, 2017-11-06:
Hello everyone, i really need help as i am new in Gazebo. 

i have cloned the PX4 Firmware source from https://github.com/PX4/Firmware in my home directory.
Then i go to ~/src/Firmware and type this in my terminal : 
$ make posix_sitl_default gazebo 
After the gazebo simulation and the iris model appeared, i open new terminal and run this command 
$ roslaunch px4 mavros.launch

The mavros has connected successfully to gazebo and i can change the mode using command 
$ rosservice call /mavros/set_mode "base_mode: 0
custom_mode: 'AUTO.LAND'"
and the result is like this 
[image]

and then i want to change to OFFBOARD mode, and the result is like this 
[image]

the mode_sent is true, but the mode is not changed to OFFBOARD mode. 

Is there anyone know how to change to offboard mode in gazebo ?

TSC21, 2017-11-06:
You cannot change to offboard without first publish a stream of setpoints. That is clear on the offboard section of the devguide.

billiepratama, 2017-11-06:
aaa i see i see.. that's work. Thankyou very much :)

closed 2017-11-06 by billiepratama