PX4/PX4-Devguide #347: Reorganise structure to organise hardware related docs

hamishwillee, 2017-11-15:
Lets reorg the DevGuide. Below is a bit of a discussion. This flows on from #345

----

With respect to "Hardware" this is a little difficult because presumably some topics cover different "domains: (?)
- "hardware you can use",
 - "configuration of specific hardware",
- Integrating hardware to use with PX4. E.g. working with UAVCan.  

Currently we MIX those up. So in [Sensors and Actuator busses](https://hamishwillee.gitbooks.io/havdevguide/content/v/collapsible_menu/en/sensor_bus/) we have 
- I2C section which explains nothing about I2C but has the config for a single specific LIDAR.
- UART section - nothing about UART, contains a doc on setting up driver for uLandingRadar
- UAVCAN - generic information about UAVCAN config and firmware update.
One  problem with that section is that no one who is interested in uLandingRadar or LIDAR will look under Sensor and Actuator busses for the information.  I am also left wondering if there is no config for I2C and UART, since we didn't talk about them. 

Anyway, my take on this:
1. Anything to do with using hardware that is config only goes into the PX4 user guide:  [sf1xx_lidar_setup.html](https://hamishwillee.gitbooks.io/havdevguide/content/v/collapsible_menu/en/sensor/sf1xx_lidar_setup.html), (https://hamishwillee.gitbooks.io/havdevguide/content/v/collapsible_menu/en/sensor/uart_ulanding_radar.html)
2. Hardware topics that remain on DevGuide should point to those sorts of docs on User guide
3. Hardware topics on DevGuide should be about development - "how to integrate with a flight controller", "how to add a sensor", "How to add an airframe"

I am not completely convinced of course - a topic about integrating with UAVCan is actually about software ... so having under Hardware might not work. THoughts?

Assuming it does, a possible structure:
- Hardware (or Hardware & Hardware Integration?)
  - Autopilot Hardware
  - Airframes
  - Data Links
    - [SiK Radio](https://hamishwillee.gitbooks.io/havdevguide/content/v/collapsible_menu/en/data_links/sik_radio.html)
    - [Telemetry Wifi](https://hamishwillee.gitbooks.io/havdevguide/content/v/collapsible_menu/en/advanced/telemetry_wifi.html) - perhaps delete, merge, or have as parent of SikRadio above?
  - Sensor and Actuator Busses
     - I2C - add information about the type of bus and integrating with it
     - UAVCAN - leave as is
     - UART - add information about the type of bus and integrating with it + config
  - Companion Computers
  - [RTK GPS](https://hamishwillee.gitbooks.io/havdevguide/content/v/collapsible_menu/en/advanced/rtk_gps.html) - move out of advanced. 

In above structure:
- [Parrot Bebop 2](https://hamishwillee.gitbooks.io/havdevguide/content/v/collapsible_menu/en/advanced/parrot_bebop.html) - could go under airframes. Even though this is a somewhat developer topic I suggest we move its content into the [user guide topic](https://docs.px4.io/en/flight_controller/bebop.html)
- [Camera Gimbal Control](https://hamishwillee.gitbooks.io/havdevguide/content/v/collapsible_menu/en/advanced/gimbal_control.html) - Under Hardware Integration this would be reasonable. Not so sure under "Hardware". 
- [Camera Trigger](https://hamishwillee.gitbooks.io/havdevguide/content/v/collapsible_menu/en/tutorials/camera_trigger.html) - probably needs to be moved into user guide.

LorenzMeier, 2017-11-15:
On reading this I think you should go ahead with your proposal. It sounds a lot more consistent than the status quo.

hamishwillee, 2018-02-23:
Closing. Everything discussed here has been done. 

There is more that can be done to pull apart advanced and tutorial sections, but all these items are covered.

closed 2018-02-23 by hamishwillee