PX4/PX4-Devguide #349: Add documentation on multicopter mixer

jlecoeur, 2017-11-15:
Following https://github.com/PX4/Firmware/pull/8063, some doc is needed.

- documentation of the .toml format used to describe multirotor geometries
- tutorial for adding a custom multirotor geometry & mixer

@hamishwillee you can assign me on this one.

hamishwillee, 2017-11-15:
Thanks @julianoes !

julianoes, 2017-11-16:
I'm confused, I don't think I have time for this until next year.

jlecoeur, 2017-11-16:
Probably a confusion between Julian and Julien !

julianoes, 2017-11-16:
Seems like it 😄

hamishwillee, 2017-11-16:
Sorry guys. Too many Juli(a|e)ns in this house!

chuong98, 2018-01-19:
Hi guys, I wonder when the documentation will be released, as I really need to add the new air frame very soon.
Thanks.

jlecoeur, 2018-01-19:
Hi @chuong98, sorry for the delay. I am working (slowly) on 6dof support in the mixers, and I was planning to document everything in one go when that is ready.

For your platform, please open an issue with a description of the rotor layout, we need the X Y Z distances between rotor and center of gravity. I will guide you from there.

chuong98, 2018-01-19:
Thanks @jlecoeur , I add an issues https://github.com/PX4/Devguide/issues/435

jlecoeur, 2018-04-16:
Related issue #511

hamishwillee, 2018-04-16:
Hi @jlecoeur 

Great to see you're keeping track of all the related material :-). Any thoughts on when you might get to first draft of the geometry docs. 

I am starting to think that we should at least mention the existence of the geometry file earlier rather than later (in adding a new airframe doc).

hamishwillee, 2018-09-04:
@jlecoeur Can we have some work on this urgently? It has been about a year and a half now. If you can't do it, can you suggest someone else who can help?

hamishwillee, 2018-09-04:
See also #511 and #435

hamishwillee, 2020-03-30:
- [ ] Note, when this is in, we need to address mixer docs for tricopter etc as per https://github.com/PX4/Devguide/pull/994#discussion_r400139445

hamishwillee, 2020-04-07:
@jlecoeur Can I assume that you're working on this?

jlecoeur, 2020-04-07:
I will give it a shot this week-end

hamishwillee, 2020-04-14:
@jlecoeur I'm still watching you :-). How did you go?

hamishwillee, 2020-04-25:
@jlecoeur @bkueng FYI. Ping.