PX4/PX4-Devguide #351: Add info in external_position_estimation doc to support fake_gps

hamishwillee, 2017-11-16:
https://github.com/PX4/Devguide/pull/350 removed "deprecated" doc on fake_gps. According to @TSC21  fake_gps is still a possibility with MAVROS -though the name changed from mocap_fake_gps.cpp to fake_gps.cpp (as it allows other sources of estimated pose besides motion capture).

The proposal is to update  https://dev.px4.io/en/ros/external_position_estimation.html to include fake_gps information. Context here: https://github.com/PX4/Devguide/pull/350#issuecomment-344861531

@julianoes - you wrote the plugin - do you have time to update docs? If not @TSC21 is "backup"

julianoes, 2017-11-17:
I don't remember any of this, sorry. Do what you think makes sense.

TSC21, 2017-11-17:
@hamishwillee assign it to me. I will take care of it as I updated this plugin in the mean time. Thanks @julianoes

julianoes, 2017-11-17:
I can't assign you @TSC21, not sure.

hamishwillee, 2017-11-18:
@julianoes Can you add @TSC21  to the https://github.com/orgs/PX4/teams/docs-team/members (requires admin). After that he can be assigned to own docs issues

julianoes, 2017-11-20:
I can 😄, done.