PX4/PX4-Devguide #353: Some minor improvements to the ubuntu_sim_ros_gazebo.sh script

cyberang3l, 2017-11-18:
The following diff provides some improvements for the script `ubuntu_sim_ros_gazebo.sh`. In particular:
  * The pip is upgraded after being installed. This removes a warning the the user has to upgrade pip.
  * The script now finds the number of physical cores and compile the eprosima libraries faster with the `make -j<number-of-cores>`.
  * The eprosima tar.gz files are removed after the compilation of the libraries has finished.
  * In my setup (KDE Neon - which is based on Ubuntu 16.04) sourcing `.bashrc` from within a script doesn't seem to be working. Sourcing directly the ROS setup.bash files is the safest way to ensure that the environment variables have been updated.
  * Before adding the source lines in the `.bashrc` file, the script is now checking that the lines do not already exist.
  * The distribution id for KDE neon (and probably other ubuntu-based distros flavors) doesn't return `ubuntu`.  This makes the command `rosdep` to return some errors:
    ```
    rosdep db
    Unsupported OS: neon
    Supported OSes are [osx, fedora, gentoo, opensuse, cygwin, slackware, ubuntu, arch, debian, rhel]
    ```
    If the `rosdep install` command returns an error, then the command is reexecuted by assuming that the distro is an ubuntu 16.04 installation: `rosdep install --from-paths src --ignore-src --rosdistro kinetic -y --os ubuntu:xenial`. A variable `unsupported_os` is updated, and the user is warned with an error message at the end of the script's execution that the script assumed that the distro is Ubuntu 16.04 and proceeded with the installation. Since the script is also called `ubuntu_sim_ros_gazebo.sh`, I believe it is fair to assume that the default OS supported by this script is ubuntu.

```diff
diff --git a/ubuntu_sim_ros_gazebo.sh b/ubuntu_sim_ros_gazebo.sh
index 6dae9d7..8f14ede 100755
--- a/ubuntu_sim_ros_gazebo.sh
+++ b/ubuntu_sim_ros_gazebo.sh
@@ -45,6 +45,7 @@ sudo apt-get install python-argparse git git-core wget zip python-empy python-to
 # Required python packages
 sudo apt-get install python-dev -y
 sudo apt-get install python-pip -y
+sudo -H pip install --upgrade pip
 sudo -H pip install pandas jinja2
 pip install pyserial
 # optional python tools
@@ -62,14 +63,17 @@ then
 else
     pushd .
     cd ~
-    wget http://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-5-0/eprosima_fastrtps-1-5-0-linux-tar-gz
+    wget http://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-5-0/eprosima_fastrtps-1-5-0-linux-tar-gz -O eprosima_fastrtps-1-5-0-linux.tar.gz
     mv eprosima_fastrtps-1-5-0-linux-tar-gz eprosima_fastrtps-1-5-0-linux.tar.gz
     tar -xzf eprosima_fastrtps-1-5-0-linux.tar.gz eProsima_FastRTPS-1.5.0-Linux/
     tar -xzf eprosima_fastrtps-1-5-0-linux.tar.gz requiredcomponents
     tar -xzf requiredcomponents/eProsima_FastCDR-1.0.7-Linux.tar.gz
-    cd eProsima_FastCDR-1.0.7-Linux; ./configure --libdir=/usr/lib; make; sudo make install
+    cpucores=$(( $(lscpu | grep Core.*per.*socket | awk -F: '{print $2}') * $(lscpu | grep Socket\(s\) | awk -F: '{print $2}') ))
+    cd eProsima_FastCDR-1.0.7-Linux; ./configure --libdir=/usr/lib; make -j$cpucores; sudo make install
     cd ..
-    cd eProsima_FastRTPS-1.5.0-Linux; ./configure --libdir=/usr/lib; make; sudo make install
+    cd eProsima_FastRTPS-1.5.0-Linux; ./configure --libdir=/usr/lib; make -j$cpucores; sudo make install
+    cd ..
+    rm -rf requiredcomponents eprosima_fastrtps-1-5-0-linux.tar.gz
     popd
 fi
 
@@ -90,8 +94,10 @@ sudo apt-get install ros-kinetic-desktop-full -y
 sudo rosdep init
 rosdep update
 ## Setup environment variables
-echo "source /opt/ros/kinetic/setup.bash" >> ~/.bashrc
-source ~/.bashrc
+rossource="source /opt/ros/kinetic/setup.bash"
+if grep -Fxq "$rossource" ~/.bashrc; then echo ROS setup.bash already in .bashrc;
+else echo "$rossource" >> ~/.bashrc; fi
+eval $rossource
 ## Get rosinstall
 sudo apt-get install python-rosinstall -y
 
@@ -114,13 +120,18 @@ rosinstall_generator mavlink | tee -a /tmp/mavros.rosinstall
 ### Setup workspace & install deps
 wstool merge -t src /tmp/mavros.rosinstall
 wstool update -t src
-rosdep install --from-paths src --ignore-src --rosdistro kinetic -y
+if ! rosdep install --from-paths src --ignore-src --rosdistro kinetic -y; then
+    # (Use echo to trim leading/trailing whitespaces from the unsupported OS name
+    unsupported_os=$(echo $(rosdep db 2>&1| grep Unsupported | awk -F: '{print $2}'))
+    rosdep install --from-paths src --ignore-src --rosdistro kinetic -y --os ubuntu:xenial
+fi
 ## Build!
 catkin build
 ## Re-source environment to reflect new packages/build environment
-echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
-source ~/.bashrc
-
+catkin_ws_source="source ~/catkin_ws/devel/setup.bash"
+if grep -Fxq "$catkin_ws_source" ~/.bashrc; then echo ROS catkin_ws setup.bash already in .bashrc; 
+else echo "$catkin_ws_source" >> ~/.bashrc; fi
+eval $catkin_ws_source
 
 # Clone PX4/Firmware
 clone_dir=~/src
@@ -135,3 +146,9 @@ else
     cd Firmware
 fi
 cd $clone_dir/Firmware
+
+if [[ ! -z $unsupported_os ]]; then
+    >&2 echo -e "\033[31mYour OS ($unsupported_os) is unsupported. Assumed an Ubuntu 16.04 installation,"
+    >&2 echo -e "and continued with the installation, but if things are not working as"
+    >&2 echo -e "expected you have been warned."
+fi
```

TSC21, 2017-11-19:
@cyberang3l you should make a PR with this. It will be easier for use to compare the diffs and to comment on those.

cyberang3l, 2017-11-19:
Here we are then: https://github.com/PX4/Devguide/pull/354 :)

closed 2017-11-24 by cyberang3l