PX4/PX4-Devguide #355: Firmware/Documentation - move into DevGuide

hamishwillee, 2017-11-19:
https://github.com/PX4/Firmware/tree/master/Documentation contains lots of developer docs. Review and determine correct place for them as per this discussion: https://github.com/PX4/Firmware/issues/8316#issuecomment-345555835

IMO even if source is right place for a single doc, that doc still needs to be in findable via the main documentation set.

dinesh286, 2017-11-22:
After running the script ubuntu_sim_nuttx.sh I m getting the following during make px4fmu-v2_default
I am using ubuntu 16.04

[6/605] Generating uORB topic headers
FAILED: cd /home/dinesh/src/Firmware/msg && /home/dinesh/anaconda3/bin/python tools/px_generate_uorb_topic_files.py --headers -f actuator_armed.msg actuator_controls.msg actuator_direct.msg actuator_outputs.msg adc_report.msg airspeed.msg att_pos_mocap.msg battery_status.msg camera_capture.msg camera_trigger.msg collision_report.msg commander_state.msg cpuload.msg debug_key_value.msg debug_value.msg debug_vect.msg differential_pressure.msg distance_sensor.msg ekf2_innovations.msg ekf2_timestamps.msg esc_report.msg esc_status.msg estimator_status.msg filtered_bottom_flow.msg follow_target.msg fw_pos_ctrl_status.msg geofence_result.msg gps_dump.msg gps_inject_data.msg hil_sensor.msg home_position.msg input_rc.msg led_control.msg log_message.msg manual_control_setpoint.msg mavlink_log.msg mc_att_ctrl_status.msg mission.msg mission_result.msg mount_orientation.msg multirotor_motor_limits.msg offboard_control_mode.msg optical_flow.msg output_pwm.msg parameter_update.msg position_setpoint.msg position_setpoint_triplet.msg power_button_state.msg pwm_input.msg qshell_req.msg rc_channels.msg rc_parameter_map.msg safety.msg satellite_info.msg sensor_accel.msg sensor_baro.msg sensor_bias.msg sensor_combined.msg sensor_correction.msg sensor_gyro.msg sensor_mag.msg sensor_preflight.msg sensor_selection.msg servorail_status.msg subsystem_info.msg system_power.msg task_stack_info.msg tecs_status.msg telemetry_status.msg test_motor.msg time_offset.msg transponder_report.msg uavcan_parameter_request.msg uavcan_parameter_value.msg ulog_stream.msg ulog_stream_ack.msg vehicle_attitude.msg vehicle_attitude_setpoint.msg vehicle_command.msg vehicle_command_ack.msg vehicle_control_mode.msg vehicle_force_setpoint.msg vehicle_global_position.msg vehicle_gps_position.msg vehicle_land_detected.msg vehicle_local_position.msg vehicle_local_position_setpoint.msg vehicle_rates_setpoint.msg vehicle_roi.msg vehicle_status.msg vehicle_status_flags.msg vtol_vehicle_status.msg wind_estimate.msg -i /home/dinesh/src/Firmware/msg -o /home/dinesh/src/Firmware/build/px4fmu-v2_default/uORB/topics -e templates/uorb -t /home/dinesh/src/Firmware/build/px4fmu-v2_default/msg/tmp/headers -q
python import error:  No module named 'em'

Required python packages not installed.

On a Debian/Ubuntu system please run:

  sudo apt-get install python-empy
  sudo pip install catkin_pkg

[6/605] Copying NuttX config px4fmu-v2 and configuring
FAILED: cd /home/dinesh/src/Firmware && make --no-print-directory px4io-v2_default
ninja: build stopped: subcommand failed.
Makefile:153: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 1

hamishwillee, 2017-11-22:
@dinesh286 Please do not put completely random support questions into existing issues. I am deleting the above. Please create a NEW issue, and include your FULL build log.

hamishwillee, 2017-12-13:
@bkueng @julianoes Can you help me work out what of [Firmware/Documentation](https://github.com/PX4/Firmware/tree/master/Documentation) is still relevant? I'd like to bring what is relevant into the devguide (updating if necessary) and deleting the rest (or at least understand why it needs to be here). 

So for each of below, are they relevant, are they accurate. If you have any ideas where they might fit into the devguide/user guide that would be a bonus.

* arming_state_machine.odg
* arming_state_machine.pdf
* code_structure_diagrams.odg
* code_structure_diagrams.pdf
* control_flow.graffle
* Doxyfile
* doxygen.README
* doxygen.sh
* dsm_bind.odt
* dsm_bind.pdf
* fixed_wing_control.odg
* flight_mode_state_machine.odg
* flight_mode_state_machine.pdf
* [fw_landing.png](https://github.com/PX4/Firmware/blob/master/Documentation/fw_landing.png) 
  * Replaced by this image: https://docs.px4.io/en/flying/fixed_wing_landing.html
* mixing_architecture.graffle
* px4_block_diagram.odg
* px4_block_diagram.pdf
* px4_general_structure.png
 * [px4_hil](https://github.com/PX4/Firmware/tree/master/Documentation/px4_hil) 
   * I think has now been superceded by the images and docs in the [devguide simulation page](https://dev.px4.io/en/simulation/#sitl-simulation-environment) and [hitl](https://dev.px4.io/en/simulation/hitl.html) page.
* rc_mode_switch.odg
* rc_mode_switch.pdf
* versionfilter.sh

bkueng, 2017-12-14:
- https://github.com/PX4/Firmware/blob/master/Documentation/control_flow.graffle
  https://github.com/PX4/Firmware/blob/master/Documentation/mixing_architecture.graffle
  don't know, can't open.
- https://github.com/PX4/Firmware/blob/master/Documentation/dsm_bind.pdf
  There might be some useful information to extract, but I think today it just works via single-click button in QGC.
- https://github.com/PX4/Firmware/blob/master/Documentation/rc_mode_switch.pdf
  Might provide some more information in addition to https://dev.px4.io/en/concept/flight_modes.html.

I think the rest can just be deleted. It's simply outdated, unused or superseded in the dev-guide. @julianoes please correct me where I'm wrong.

julianoes, 2017-12-14:
Yer, I'm pretty sure it's old and outdated.

hamishwillee, 2017-12-20:
@bkueng There is a free chrome omnigraffle editor [here](https://chrome.google.com/webstore/detail/free-omnigraffle-graffle/ahlhlpmldjfennoppdkhpnljgihpnnpj/related). The images you can't see are:

control_flow.graffle:
[image]

mixing_architecture.graffle - possibly broken. 

[image]

The second doc should IMO be deleted. Even if this does render correctly in omnigraffle it isn't useful to most of our team.

Thoughts on the first doc?

hamishwillee, 2017-12-20:
@bkueng Regards

hamishwillee, 2017-12-20:
@bkueng Regarding  [dsm_bind.pdf](https://github.com/PX4/Firmware/blob/master/Documentation/dsm_bind.pdf) 

> There might be some useful information to extract, but I think today it just works via single-click button in QGC.

Yes, today this is easily done through QGC as [documented here](https://docs.px4.io/en/config/radio.html#spektrum_bind). 

I propose this is deleted because it is completely out of date for any supported Pixhawk series board (is all pre 3DR Pixhawk when there was a separate board). I guess some of the parameters and generic information about connecting to Spektrum might be interesting if you wanted to add this binding support for some other board, but working out what is still relevant would be very hard.

hamishwillee, 2017-12-20:
@bkueng re [rc_mode_switch.pdf](https://github.com/PX4/Firmware/blob/master/Documentation/rc_mode_switch.pdf)

Who could help with the integration into flight modes? These diagrams are meaningless to me - I don't know what a switch is - who would use them, and when/why? Probably need at least some words (each) explaining why they are useful.

bkueng, 2017-12-20:
> Thoughts on the first doc?

Some of it is outdated, the rest is on the new architecture diagram. I'd say this can be removed.

> Who could help with the integration into flight modes? These diagrams are meaningless to me - I don't know what a switch is - who would use them, and when/why? Probably need at least some words (each) explaining why they are useful.

Well the swich is just the one(s) on your RC controller. What I know is that there is an old hierachical logic for switching modes where you use multiple switches, and a new one with just a single multi-position switch (default in QGC, there's a button `Use Multi Channel Mode Selection` for the old behavior, but I've just used the single switch).

hamishwillee, 2017-12-20:
Thank you. Lorenz has stated we can delete everything except the Doxygen files.

closed 2017-12-21 by hamishwillee