PX4/PX4-Devguide #357: Simulation UDP port description is a bit confusing

cyberang3l, 2017-11-20:
In this page: https://dev.px4.io/en/simulation/, the section `Standard MAVLink UDP Ports` has the following content:

```
PX4 defines a number of standard UDP broadcast ports that ground control stations (e.g. QGroundControl, Offboard APIs and simulators) can listen to in order to connect to the autopilot:

Offboard APIs: 14540
Ground Control Stations: 14550 (QGroundControl automatically connects to PX4 broadcasting on this port).
Simulators: 14560
```

Technically, this is not precise.

As I understand after some further investigation, the simulators do not listen to port `14560`. They listen to a randomly chosen port. If one executes the `ss -upnl4` command when running a gazebo simulation, they will see something like this:

```bash
$ ss -upnl4
State      Recv-Q Send-Q          Local Address:Port                         Peer Address:Port              
UNCONN     0      0                           *:14556                                   *:*                   users:(("px4",pid=3992,fd=7))
UNCONN     0      0                           *:14557                                   *:*                   users:(("px4",pid=3992,fd=8))
UNCONN     0      0                           *:14560                                   *:*                   users:(("px4",pid=3992,fd=5))
UNCONN     0      0                           *:631                                     *:*                  
UNCONN     0      0                           *:42044                                   *:*                   users:(("gzserver",pid=3963,fd=34))
UNCONN     0      0                           *:35909                                   *:*                  
UNCONN     0      0                           *:5353                                    *:*
```

It is obvious from here that PX4 is listening to 14556, 14557 and 14560, and the simulator is listening to a random port (in this case 42044).

A tcpdump also shows that it is PX4 that is using the src ports 14556 and 14557 to broadcast data to the dst ports 14550 and 14540 respectively, but it is the simulator that is using the random src port (42044 in this example) to broadcast data to the port 14560.

```bash
$ sudo tcpdump -i any -n -c5 'udp and port 14556'
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
18:49:30.656139 IP 127.0.0.1.14556 > 127.0.0.1.14550: UDP, length 70
18:49:30.660830 IP 127.0.0.1.14556 > 127.0.0.1.14550: UDP, length 70
18:49:30.667071 IP 127.0.0.1.14556 > 127.0.0.1.14550: UDP, length 70
18:49:30.667080 IP 127.0.0.1.14556 > 127.0.0.1.14550: UDP, length 36
18:49:30.667084 IP 127.0.0.1.14556 > 127.0.0.1.14550: UDP, length 36
5 packets captured
24 packets received by filter
0 packets dropped by kernel

$ sudo tcpdump -i any -n -c5 'udp and port 14557'
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
18:49:33.037551 IP 127.0.0.1.14557 > 127.0.0.1.14540: UDP, length 36
18:49:33.048615 IP 127.0.0.1.14557 > 127.0.0.1.14540: UDP, length 70
18:49:33.048622 IP 127.0.0.1.14557 > 127.0.0.1.14540: UDP, length 36
18:49:33.048625 IP 127.0.0.1.14557 > 127.0.0.1.14540: UDP, length 40
18:49:33.058077 IP 127.0.0.1.14557 > 127.0.0.1.14540: UDP, length 36
5 packets captured
12 packets received by filter
1 packet dropped by kernel

$ sudo tcpdump -i any -n -c5 'udp and port 14560'
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
18:49:36.082381 IP 127.0.0.1.42044 > 127.0.0.1.14560: UDP, length 72
18:49:36.082401 IP 127.0.0.1.42044 > 127.0.0.1.14560: UDP, length 72
18:49:36.084372 IP 127.0.0.1.42044 > 127.0.0.1.14560: UDP, length 72
18:49:36.084391 IP 127.0.0.1.42044 > 127.0.0.1.14560: UDP, length 72
18:49:36.086380 IP 127.0.0.1.42044 > 127.0.0.1.14560: UDP, length 72
5 packets captured
12 packets received by filter
0 packets dropped by kernel
```

Moreover, although technically data is broadcasted from port 14556, 14557 and the random simulator port towards the ports 14550, 14540 and 14560 respectively, these ports are not strictly *src* and *dst* ports. They are used for bidirectional communication, thus, they can be both *src* and *dst* ports depending on the direction of the packets.

As you can see from the following tcpdump, when I sent the command `commander takeoff` to lift the drone in the air, suddenly PX4 broadcasts data towards the simulator (makes sense). In this case the port 14560 is the *src* port (you can see it from the tcpdump filter as well. I have asked to filter the src port 14560):

```bash
$ sudo tcpdump -i any -n -c5 'udp and src port 14560'
[sudo] password for vangelis: 
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
19:05:30.935813 IP 127.0.0.1.14560 > 127.0.0.1.42044: UDP, length 89
19:05:30.940887 IP 127.0.0.1.14560 > 127.0.0.1.42044: UDP, length 89
19:05:30.945819 IP 127.0.0.1.14560 > 127.0.0.1.42044: UDP, length 89
19:05:30.948336 IP 127.0.0.1.14560 > 127.0.0.1.42044: UDP, length 89
19:05:30.953305 IP 127.0.0.1.14560 > 127.0.0.1.42044: UDP, length 89
5 packets captured
10 packets received by filter
0 packets dropped by kernel
```

Why I started talking about *src* and *dst* ports then? Because of this figure that got me confused even further:

[image]

I believe what would make this figure a bit more clear is to remove the src/dst labels for each of the ports, and associate the ports to the correct components.

E.g. in a standard service like DNS, one would probably draw something like this:
```
┏━━━━━━┓                   ┏━━━━━━┓
┃server┃53 <------>randPort┃client┃
┗━━━━━━┛                   ┗━━━━━━┛
```
Then it is clear that the server is using port 53, and the client is using a random port.

I would be happy to get some feedback in this whole topic.

I also drawn the following figure that I believe makes things a bit more clear. Feel free to use it if you think the same (I know it's not the same style like the rest of the figures, but some color doesn't hurt. I used color-blind friendly colors).
[image]

hamishwillee, 2017-11-21:
@cyberang3l  Thanks for your comments. My understanding of UDP is that each message will have a src and dst port listed. If something broadcasts a message it includes its src port which listeners must reply to. In this case the  diagram shows src and dst  with *respect to the PX4* .  So for offboard PX4 sends to 14540, specifying a return (src) address of 14557.

Based on your research it sounds like the existing diagram is miss-labeled for the simulator - the PX4 src is 14560 from PX4 point of view (not the dst). So PX4 will listen on 14560 and set that as the src to the simulator. This also indicates that the Simulator will initiate the comms, since in the first case PX4 has no idea what port the simulator will listen on. 

Anyway, you're absolutely right that this diagram is not helpful. It is much better to mark the listening ends.

1. I'd like to take your diagram, but then the style would be inconsistent with the [HITL diagram](https://dev.px4.io/en/simulation/#hitl-simulation-environment). I'll update mine to look like yours.

2. With respect to the text "PX4 defines a number of standard UDP broadcast ports...", would you be happy with the following text?

----

## Standard MAVLink UDP Ports

PX4 defines a number of standard UDP ports that ground control stations (e.g. *QGroundControl*, Offboard APIs, simulator APIs) can use to connect to the autopilot:

- Offboard APIs listen to: 14540
- Ground Control Stations listen to: 14550  (*QGroundControl* automatically connects to PX4 broadcasting on this port).
- Simulators broadcast to: 14560

> **Note** The ports for the GCS and offboard APIs are set in configuration files, while the simulator broadcast port is hard-coded in the simulation MAVLink module. 

## SITL Simulation Environment

The diagram below shows a typical SITL simulation environment for any of the supported simulators. The different parts of the system connect via UDP, and can be run on either the same computer or another computer on the same network.

* PX4 uses a simulation-specific module to listen on UDP port 14560. Simulators connect to this port, then exchange information using the [Simulator MAVLink API](#simulator-mavlink-api) described above. SITL and the simulator can run on either the same computer or different computers on the same network.
* PX4 uses the normal MAVLink module to connect to  GroundStations (listening on port 14550) and external developer APIs like DroneCore or ROS (listening on port 14540).
* A serial connection is used to connect Joystick/Gamepad hardware via *QGroundControl*.

UPDATED IMAGE HERE (then continue as before....)

cyberang3l, 2017-11-21:
The suggested version looks much better, but there is still something confusing. Here are some more of my thoughts:

The first sentence writes: `PX4 defines a number of standard UDP ports that ground control stations (e.g. QGroundControl, Offboard APIs, simulator APIs) can use to connect to the autopilot`. Then the bullet points start with `Offboard APIs listen to: 14540, GCS listen...` etc.

So since PX4 `defines`, why we start talking about who `listens`? (expected to listen would be more correct here if we go with this wording).

Moreover, the title of this section is `Standard MAVLink UDP Ports`. Are these ports defined by the MAVLink protocol or PX4? I believe it is the former that has some well-defined default ports for communication.

Just to conclude with my thoughts. Another *thing* that contributed towards my initial confusion is also the way the word `broadcasting` is used. In general, when you have a service that broadcasts information, usually you have several clients that receive information from the broadcaster and the clients do not have a bidirectional communication, i.e. do not send data back to the broadcaster. This is not the case with PX4.

Technically, PX4 indeed broadcasts information to these ports (even if there is no one listening, it will still broadcast the data), but it also uses the same ports to receive replies from whoever listens on the other side. If there are more than one remote entities listening to the same port and both reply independently, things become very messy. I experienced lots of that in my journey to understand how to achieve a multi-drone simulation.

How about something like this (I find Note 2 quite important in order to clarify the concept of `broadcasting` in the context of PX4):

-------------------
## Standard MAVlink UDP ports

By default, PX4 uses the well-defined MAVLink UDP ports for communication with ground control stations (e.g. QGroundControl), Offboard APIs (e.g. MAVROS) and simulator APIs (e.g. Gazebo). These ports are:

* Port **14540** is used for communication with offboard APIs. PX4 broadcasts data towards this port, and Offboard APIs are expected to listen.
* Port **14550** is used for communication with ground control stations. PX4 broadcasts data towards this port, and GCS are expected to listen. QGroundControl listens to this port by default.
* Port **14560** is used for communication with simulators. PX4 listens to this port, and simulators are expected to initiate the communication by broadcasting data to this port.

> Note 1: The ports for the GCS and offboard APIs are set in configuration files, while the simulator broadcast port is hard-coded in the simulation MAVLink module.
> Note 2: The aforementioned ports are used for bidirectional communication, and although technically one side starts broadcasting data towards the *unknown* other side(s), the communication doesn't intend to be *one-to-many*. If for example you have multiple independent offboard API's communicating with a single PX4 instance that transmits data towards the port 14540, things will become messy.
-------------------

Section `SITL Simulation Environment` as described and with the updated figure will be very clear.

I also redesigned the HITL figure to match the style of my previous figure. You can use them both if you want (I also have the Inkscape SVG files that I can send if needed).

[image]

hamishwillee, 2017-11-22:
Thanks for your clarifications. 
Thanks!

1. Can I have that copy of the updated HITL diagram (2/3 diags on the page are above, but not the HITL one)
2. 
   > Moreover, the title of this section is Standard MAVLink UDP Ports. Are these ports defined by the MAVLink protocol or PX4? I believe it is the former that has some well-defined default ports for communication.

   PX4. So perhaps "Standard PX4 MAVLink UDP Ports" ?

3. I'm not sure Note 2 is desirable.  AFAIK MAVLink and the comms themselves are OK with many offboard APIs, provided that they each set a unique system ID.  What is true is that we don't properly support multiple offboard APIs - but that is an issue of PX4 not (yet) handling prioritization, and we plan to address that in coming releases.

cyberang3l, 2017-11-22:
I see. Then I would not call these ports `Standard`, but `Default`. The word standard leads me to think about something which is standardized. The `standard` is the communication protocol in our case. So if the communication protocol defines or provides some guidelines for defaults ports to use, then this is considered to be the `standard`. If there is no clarification on that part from the protocol itself, then the ports are simply the `default` ports that the PX4 software chooses for communication.

Think about for instance the TCP port 80 that is the standard port used by the HTTP protocol. Port 80 is a standard, well-defined port. If I write an application that uses port 90 for HTTP communication, then this is the default for my application but not the standard. If there was no standard port defined for the HTTP protocol (by its RFC), then I could not call `standard` the port 90 that I chose to use in my application. Another application could have chosen port 100 for example.

I'm sorry for these long discussions and word nitpicking. I don't want to be the a *weirdo*, but it is this minor wording and slightly inaccurate figures that misled me and made it harder for me to get to understand how things work and different components communicate with each other.

As for the HITL figure, I completely missed that :blush: . Silly me. And I was wondering why you referred to the figure right above the SITL figure as the `HITL` :smile: .

I will design the `HITL` figure as well (now that I managed to spot it).

I just need one clarification. What is the `STL2` block? This block exists in both the SITL and the HITL figures (and I have also mislabeled the STL2 block in the SITL figure as I see. I'll correct that).

cyberang3l, 2017-11-22:
One more question about the HITL figure.
Is there any semantic meaning for the solid/dotted arrows?

cyberang3l, 2017-11-22:
Here is all three figures redesigned.

[image]

[image]

[image]

hamishwillee, 2017-11-22:
Hi @cyberang3l 

The ports are kind of a "defacto" standard - used by other autopilots too. That said, I agree that default is a better word because even though it would be silly to modify them, they can be changed. 

> I'm sorry for these long discussions and word nitpicking. I don't want to be the a weirdo, but it is this minor wording and slightly inaccurate figures that misled me and made it harder for me to get to understand how things work and different components communicate with each other.

It is incredibly useful to have first time readers comment on this, because a lot of the review is from very busy people who tend to rubber stamp. So while I find it irritating that this wasn't perfect, keep on going :-)

Thanks for the new diagrams!

> I just need one clarification. What is the STL2 block? This block exists in both the SITL and the HITL figures (and I have also mislabeled the STL2 block in the SITL figure as I see. I'll correct that).

It is a typographic error :-( - should be [SDL2](https://www.libsdl.org/index.php), the library that handles joysticks. This block should be removed in all diagrams as it doesn't add anything to the discussion.
Sorry, but can I ask for fix on the HITL/SITL in line with that.

You also asked about the dashed vs solid lines. I was inconsistent there too. The solid was supposed to be for serial and dashed for wireless.

hamishwillee, 2017-11-23:
So the changes are "started" in #363.  I'll add the images there when you get them. Feel free to comment further.

cyberang3l, 2017-11-23:
Black dashed lines: Internal component communication.
Purple solid lines: Serial communication.
Red dashed lines: MAVLink UDP communication. Can be wireless or not.

[image]

[image]

[image]

hamishwillee, 2017-11-23:
Yes. Very nice. See comments in #363 . Closing.

closed 2017-11-23 by hamishwillee