PX4/PX4-Devguide #359: Proposed improvements to build scripts

hamishwillee, 2017-11-22:
The build scripts are relatively convoluted/brittle/annoying. Issues for discussion/improvement.
- They put binaries in home directories and update the paths
- Rely on random PPAs that could break underneath us
- mix apt and pip for getting python dependencies

Some of the dependencies can be obtained from reliable/default packaging in 16.04 - e.g. ninja, cmake. Where possible we should do so. 

Open question is what ubuntu versions we support via scripts and the docs. IMO we should support the  current LTS for both, but provide links on github to older versions of the docs.

hamishwillee, 2017-11-23:
@dagar For common build script I removed the line below so we use default repos
```
#sudo add-apt-repository ppa:george-edison55/cmake-3.x -y
```
For some reason this then meant that `pip install pyulog` and `pyserial` failed, so I replaced them with 
```
sudo -H pip install pyserial pyulog
```
Last of all I used apt-get to fetch ninja-build. That is v1.5 while we were using 1.6 before.

Anyway, both nuttx and jmavsim build fine. Is it worth making this change "real"? It is certainly in-line with your point that we should avoid specials repos.

dagar, 2017-12-06:
Sounds good to me. With ubuntu 16.04 CI uses whichever ninja and cmake is packaged.