PX4/PX4-Devguide #369: Wifi pages need to be merged/moved

hamishwillee, 2017-11-23:
Devguide has 3 telemetry pages, most of which are actually user guide.
* [SiK Radio · PX4 Developer Guide](https://dev.px4.io/en/data_links/sik_radio.html) 
* [Telemetry · PX4 Developer Guide](https://dev.px4.io/en/tutorials/telemetry.html)
* [Telemetry Wifi · PX4 Developer Guide](https://dev.px4.io/en/advanced/telemetry_wifi.html)

These need to be merged into one *developer* topic and any user information pushed to the user guide.

hamishwillee, 2017-11-23:
@bkueng Is it correct that SiK radio is telemetry over normal radio frequencies? Wifi telemetry is the same thing but using wifi frequencies/standards for the connection?
Is there any equivalent to SiK for wifi? Ie a standardised radio design? Are there any wifi radios we can talk about other than the 3DR one?

bkueng, 2017-11-24:
> Is it correct that SiK radio is telemetry over normal radio frequencies? Wifi telemetry is the same thing but using wifi frequencies/standards for the connection?

Yes & yes. For radio, there are 2 commonly used frequencies: 915Mhz and 433Mhz.

> Is there any equivalent to SiK for wifi? Ie a standardised radio design? Are there any wifi radios we can talk about other than the 3DR one?

Not that I'm aware of.

hamishwillee, 2017-12-04:
Thanks. I've ported the docs to the user guide: https://github.com/PX4/px4_user_guide/pull/106

Question is, what to have in the Devguide? It makes sense to just have a simple doc like the entry point in the user guide. It would be great to also have instructions for anyone wanting to create their own new telemetry system. Can you help or suggest someone to help provide direction on that? Or do you think this will never happen and is pointless?

bkueng, 2017-12-04:
Hmm hard to say. I see two cases:
- mavlink-based telemetry (prefered). Pretty much all you need is to provide a UART interface with the proper voltage levels/connector, then attach it to the telemetry port of a pixhawk-based HW and you're done.
- integration of a third-party telemetry system with their own protocol (for example FrSky). It's hard to provide general guidance here since it might not even use UART, or uses different connectors, etc.

hamishwillee, 2017-12-05:
Done!

closed 2017-12-05 by hamishwillee