PX4/PX4-Devguide #379: Proposal for updating driver information

hamishwillee, 2017-12-05:
At the moment our guidance is just wrong - "use the DriverFramework" (apparently there is a plan to instead plan to get the existing NuttX drivers working on other platforms?)

The update plan would be to update https://dev.px4.io/en/middleware/drivers.html
- Move the "bringing up the platform" section into the [porting guide](https://dev.px4.io/en/debug/porting-guide.html )
- Explains what the driverids actually are
- Remove the link to DriverFramework assuming you aren't using this. If you have used it "a bit" we'd just add a note that is deprecated.

Then we'd add some general guidance:
- What are the base classes, and what do they do for you? For example, there looks like base classes for ports (e.g. i2C). There also look to be base classes for functionality - e.g. baro, airspeed, whatever. 
- How you talk to the port(s)
- How you talk to PX4 (uorb)
- "What" do you need to talk to PX4? ie presumably there is a driver on the UART. If you are using the UART you have to communicate with that. How is it done. 
- How you add the driver to a firmware config
- How/when you start the driver. Can it be plug'n play? Can it make itself auto-detect?
- IF a driver does require config, how to do that - ie how to provide param say to tell system that new baro "fred" is on a particular UART or on I2C or whatever.
- What test code must be provided. 
- Anything else?
add some good examples to point to for those points.

In addition, I want to cross link to other docs when/if more targetted information can be provided. Eg.. https://dev.px4.io/en/sensor_bus/i2c.html

hamishwillee, 2017-12-05:
@dagar Need your advice on this :-)

hamishwillee, 2017-12-05:
Additional notes/discussion: http://discuss.px4.io/t/how-to-write-a-driver-code-for-external-devices-i2c-messages-from-arduino-to-pixhawk/5011

dagar, 2017-12-05:
The full picture of where we think we're headed for driver unification and simplification is TBD, but there's still more than enough detail we can fill in for now that won't change all that much.

dagar, 2017-12-05:
@davids5 fyi

hamishwillee, 2017-12-05:
Awesome. Anything you gentlemen can provide to answer the above would be useful.

hamishwillee, 2017-12-13:
Driver doc moved in https://github.com/PX4/Devguide/pull/389

hamishwillee, 2017-12-13:
https://github.com/PX4/Devguide/pull/390 does minimal updates to remove suggestion users should port to DriverFramework and provide minimal "how to get started"

hamishwillee, 2018-02-11:
Sensor drivers (and perhaps drivers in general) should include params as part of their definition. See https://github.com/PX4/Firmware/pull/8869/files
This is something to mention in driver guide.