PX4/PX4-Devguide #380: Update Archlinux instructions/script

hamishwillee, 2017-12-06:
A working script can be found in the docker setup. See https://github.com/PX4/containers/pull/84

hamishwillee, 2017-12-06:
@jlecoeur I just tried out the archlinux script from https://github.com/PX4/containers/pull/84 in a fresh VM of very latest Archlinux ISO. 

To install did:
1. Url shortener on https://raw.githubusercontent.com/PX4/containers/master/docker/px4-dev/scripts/archlinux_install_script.sh to https://goo.gl/vJYAep
2. Entered:
   ```
   wget https://goo.gl/vJYAep -O px4_setup.sh
   source ./px4_setup.sh
   ```
Got errors below. Suggestions/why does docker work, but the ISO from archlinux not?
[image]

jlecoeur, 2017-12-06:
It looks like pacman did not install make and python-pip. Can you paste the full output?
Are you running the script as root?

hamishwillee, 2017-12-06:
Hmmm. I think the problem is that archlinux requires partitioning and the like to set up initial drives. I'm going to try follow this http://wideaperture.net/blog/?p=4503
Problem though, is that this will mean that our existing instructions aren't going to "really" work out of the box.

jlecoeur, 2017-12-06:
Yes, default arch is really minimal diy system. I actually use antergos, which is archlinux + a nice installer.
I think that the script should not handle arch set-up, and should assume a functional system.

hamishwillee, 2017-12-28:
This is partially fixed by https://github.com/PX4/Devguide/pull/382

The instructions aren't great though - they use GCC from package manager (the "official instructions" didn't work) and do not build RTPS. Both those need to be fixed and testing needs to be done as standard on ArchLinux via docker. For now I added disclaimers on the docs.