PX4/PX4-Devguide #384: Document multi-sitl usage with Gazebo (but not mandating ROS)

LorenzMeier, 2017-12-09:
@hamishwillee @dagar we need to make our ability to run multiple SITL instances with and without ROS.

We have this documentation, which requires ROS:
https://dev.px4.io/en/simulation/multi-vehicle-simulation.html

And this repo contains some great ideas how to support this with and without ROS:
https://github.com/acsl-mipt/multiple-sitl/issues/3

hamishwillee, 2018-01-04:
Also info here: https://github.com/PX4/Firmware/pull/7235

weiminshen99, 2018-02-27:
Hi, 
I can run multi SITL in one gazebo using ROS just fine, but I still have trouble to do so without ROS. I have read many posts on the topic (e.g. #7235), but still lack a good understanding, so I did the following experiments:
1. I started a gazebo world with four iris in it, and they are named iris, iris_0, iris_1, iris_2.
2. now, I want to do one SITL at a time (incrementally) as follows:
    2.1. create a new PX4 SITL 
    2.2. connect the new SITL to gazebo
    2.3. bind the new SITL with one of four iris as I wish
Inspired by the various scripts I found, I use the following line command to do 2:
% $src_dir/build/posix_sitl_default/px4 $src_dir $src_dir/posix-configs/SITL/init/ekf2/iris_1
% $src_dir/build/posix_sitl_default/px4 $src_dir $src_dir/posix-configs/SITL/init/ekf2/iris_2

The first line above did well and iris_1 uses port 14560 and connects gazebo and binds the name without error. The second line has trouble. For iris_2, I tried port 14560, 14562, 14564, etc, and none of them works. 
--- If iris_2 uses 14560, then the existing gazebo will connect but PX4 has an error "WARN  [simulator] bind failed"
--- If iris_2 uses 14562, then the existing gazebo will NOT connect as if it is not there "INFO  [simulator] Waiting for initial data on UDP port 14562. Please start the flight simulator to proceed."

The files iris_1 and iris_2 are modified for this purpose, and I specified the port with the following line:
simulator start -s -u 1456x

Could someone please tell me how to proceed. Many thanks!

hamishwillee, 2019-07-22:
@lamping7 Do you think you might be able to try out this setup for multi-sitl usage with Gazebo but not mandating ROS - ie just follow instructions here: https://github.com/acsl-mipt/multiple-sitl/tree/for-px4-pr on current firmware and confirm they all work?

We do have multi-vehicle with JMAVSim now, which allows simulation without the overhead of ROS, but I still think that some people might like this. I figure we could just add another section to http://dev.px4.io/master/en/simulation/multi-vehicle-simulation.html#multiple-vehicles-using-sdf-models

If you don't have bandwidth to help with this (or you see other issues), please let me know.

lamping7, 2019-07-22:
With the changes to startup and rCS that occurred between then and now, without running it, I don't think that ruby script is compatible anymore. Also, rCS files don't need to be generated anymore (just use the `-i` argument). A script could be made that sets up the sdf or xacro model's simulation port properly like we do in the ROS launch files. But none of this actually works until we de-conflict rootfs (the params and dataman) for each sitl instance. I submitted a quick-fix PR for this problem, but it is not in yet.

hamishwillee, 2019-07-23:
@lamping7 Thank you. I'm going to make an executive decision that this issue be closed as we have multi-vehicle support without ROS via JMAVSim. 

Obviously jmavsim only supports multicopters, so if someone wants to get Gazebo-only multi-vehicle simulation working (again) then we'll happily take it.

closed 2019-07-23 by hamishwillee