PX4/PX4-Devguide #387: Document other supported bus types

hamishwillee, 2017-12-13:
We have either docs or placeholders for UAVCAN, I2C, UART. What other types of bus do we have?
SPI? What else?

-----
Generally we should follow the pattern for these of [I2C](https://dev.px4.io/en/sensor_bus/i2c.html):
- What is bus. 
- How does it differ from other alternatives
- What is it good for - ie what peripherals, on main board or off board
- Whether it supports multiple devices on bus or just one
- Standard connectors
- How connected for standard boards.
- Any issues of powering
- Any limitations of our support on the bus - ie do we have only one port supported on a device.

Then developer integration questions
- How do we integrate with it
- What does the driver need to derive from
- How is driver included in hardware
- How is driver enabled.
- How are settings handled - ie typical params associated with this
- If there are multiple ports for this bus, how do you specify which one device is attached to.

hamishwillee, 2017-12-13:
@bkueng This is a bit of a placeholder for further work. At this point I was wondering if you could just let me know what busses we have that are not documented.

bkueng, 2017-12-13:
That looks pretty good. I don't have any other bus. (PWM can be used as input/output as well, but it's not a bus).

hamishwillee, 2017-12-13:
Do you think we should perhaps then change section to *Sensor and Actuator I/O*? Or are they better separate?

bkueng, 2017-12-14:
Yes, I think it makes sense to combine them.

hamishwillee, 2017-12-20:
> Yes, I think it makes sense to combine them.

Thanks - fixed  in #395

hamishwillee, 2017-12-20:
Notes for when documenting PWM Ports (From @bkueng here: https://github.com/PX4/Devguide/pull/395#discussion_r157917183 )

Re "also the main PWM outputs."
> What I/O is typically run on this port? Motor control obviously and other actuators (e.g. parachutes). But what about input?

Lidar Lite, or some ultrasonic sensors.

> What are the mechanics of integrating a sensor on these ports. Specifically, say I want to add X sensor to an AUX port, am I again writing a device driver that reads the inputs and outputs uORB topics?

Basically yes, but I think it's somewhat involved. I think it's the easiest to point to an existing driver

> What about actuators - say I want to open a brand new type of gripper?

actuators are simpler: you choose an actuator controls topic and then configure the mixer accordingly (I think it's documented on the mixer page).