PX4/PX4-Devguide #391: Doc Bug: Interfacing to ROS · PX4 Developer Guide

jannsta1, 2017-12-14:
DESCRIBE PROBLEM WITH DOCS HERE

I believe the line ```source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default``` in the **Launching Gazebo with ROS wrappers** section of the referenced page should be modified to ```source Tools/setup_gazebo.bash $(pwd) $(pwd)/posix_sitl_default```. At least in the branch I'm using (v1.6.5) the posix_sitl_default folder appears to be placed directly in the Firmware folder.

Perhaps this is supposed to be implied but took me quite a while to figure out why I couldn't get the mavros_posix_sitl.launch file to work properly.

Bug Page: [Interfacing to ROS · PX4 Developer Guide](https://dev.px4.io/en/simulation/ros_interface.html)

jannsta1, 2017-12-14:
My apologies, ```source Tools/setup_gazebo.bash $(pwd) $(pwd)/build_posix_sitl_default``` is what works in my environment.

hamishwillee, 2017-12-14:
Thanks. I believe you're say we can close this :-).

> Perhaps this is supposed to be implied but took me quite a while to figure out why I couldn't get the mavros_posix_sitl.launch file to work properly.

Can you offer any advice how we might make debugging this easier for people?

closed 2017-12-14 by hamishwillee

jannsta1, 2017-12-14:
I’m not sure of the mechanism that defines where the build code is placed so I’m not sure if there is a hard and fast rule for where it will be? Maybe add a comment on that line that the path may be .../“build_posix_sitl_default”?

Also, I found that setting the verbose flag on the roslaunch file helped me diagnose the problem. Perhaps adding a note about this would be a helpful general comment. Otherwise you don’t get any warnings that the motor plugins haven’t been found. I’d be tempted to add that flag into the code snippet.

Another nice to have would be how to integrate customised gazebo quadcopter models into the code base. Took me a bit to figure that out and I’m still not sure that I’m following best practices.

mrivi, 2017-12-15:
@jannsta1 ` ../build_posix_sitl_default` is the old path to the built folder. It got changed to `../build/posix_sitl_default` a couple of months ago.

jannsta1, 2017-12-15:
Ah OK, thanks @mrivi good to know.

hamishwillee, 2017-12-18:
Thanks @jannsta1 - I added a note about the flag in #393 - see https://github.com/PX4/Devguide/pull/393/commits/80951e4f27b8593385c9cb6c0ed5a31021496858

I'll look at your other points it that issue too.