PX4/PX4-Devguide #398: Missing Topic: How to integrate customised Gazebo copter models into the code base

hamishwillee, 2017-12-21:
From #391 "Another nice to have would be how to integrate customised gazebo quadcopter models into the code base. Took me a bit to figure that out and I’m still not sure that I’m following best practices."

* Do we have advice on that?
* Who is best person to ask how that should be done
* Where best place to document that?
* What are the sections/sub questions that would need to be answered for this - e.g.  how to dimension the motor models

hamishwillee, 2017-12-21:
@jannsta1 ,@TSC21 suggested we open a standalone issue for your question in #391. Do you think you might be able to share the approach that you used to seed our documentation efforts? My experience is that it is much easier to get feedback if you say "I did it this way, is that right" than to ask for someone to start the docs from scratch.

jannsta1, 2017-12-21:
Hello @hamishwillee, perhaps this is a useful starting point?:

# Creating custom gazebo models for use with px4 sitl

There are 2 situations where this is useful. Specifying a new physical model and/or specifying a new px4 software environment. These can be achieved independently but both cases require a rebuild of the posix-configs build.

## Incorporating a custom physics model

- Guides for creating compatable models can be found elsewhere (e.g. on the [rotors](https://github.com/ethz-asl/rotors_simulator/wiki) website. Looking at the existing models in /Firmware/Tools/sitl_gazebo/models is also a good starting point).

- Once created, the custom model is added to /Firmware/Tools/sitl_gazebo/models along with any related mesh and .sdf files.

- we also must create a world in /Firmware/Tools/sitl_gazebo/worlds
  - it appears like this has to match the name of the model??

## Defining the px4 modules / parameters

-  a config file must be added to; Firmware/posix-configs/SITL/init/**lpe** (copying and renaming an existing config is a good way to start).
  - nb **lpe** is an example, this must be changed to the desired build type.
- this config file is where sitl modules and initial px4 parameters are specified. Again these files are relatively self-explanatory so looking at existing files in this location a good way of understanding how they work.

## Building with the custom model

- Once created or modified, the whole posix-configs build must be repeated in order for the model to become available.

- the model target name must be added to: Firmware/src/firmware/posix/sitl_target.cmake
  - it is added to the following line (currently line 84 in release 1.6.5):

```
set(models none iris iris_opt_flow ... rover hippocampus <add your model name here>)
```
- The model is parsed by /Firmware/Tools/sitl_run.sh so it is worth examining this bash script for an understanding of what happens in the build process.

## Comments:

 - It would be beneficial if the step of adding to the list of models in the .cmake file was not necessary.
 - I think the solution to this as it stands is to fork rather than clone the repo but this involves merging/rebasing for each update.
 - More flexible .world file usage may be of interest for some users (those not launching gazebo and mavros seperately).

hamishwillee, 2017-12-21:
Yes @jannsta1 - definitely. I will have a proper look at this after Xmas (away today). Thank you!

@TSC21 in the meantime, can you look at the advice above and comment 
- what else needs to be said/what would you like to say
- are there better ways to do the above
- can we address any of those comments?
- is there anyone else you suggest we involve in this discussion?

hamishwillee, 2017-12-27:
@TSC21 Appreciate you may be off on holiday. Do you think you could ETA when you might have time to discuss this/ and suggest others that might be interested in being involved?

TSC21, 2017-12-27:
@hamishwillee Sorry I missed the above. I will have write some comment soon.

hamishwillee, 2018-01-03:
Thanks @TSC21 - I'm watching you :-)

TSC21, 2018-01-03:
Yeah just cleaning and tightening some stuff on `sitl_gazebo` now and will get back to docs after, including this.

mamariomiamo, 2018-07-11:
is there any update with this issue? :)

hamishwillee, 2018-07-12:
> is there any update with this issue? :)

@TSC21  ?

kuiskas, 2018-09-26:
Hi,
did the procedure change? I have been working in an older version, but now I wanted to upgrade to 1.8 and there is no ekf2 or  lpe config files... in the step  **Defining the px4 modules / parameters** as it used to be under posix-configs/init.
Can somebody tell me how to implement now the px4 parameters for sitl?
thanks in advance

kuiskas, 2018-09-26:
I found it (I believe) it is moved to ROMFS/px4fmu_common/init.d-posix, is that right?

hamishwillee, 2018-09-27:
@kuiskas I think so. The recent changes for startup in posix and nuttx are captured in [System Startup](https://dev.px4.io/en/concept/system_startup.html#system-startup)

lamping7, 2018-09-28:
Some of that above it's not necessary depending on what is being done or what the objective is. LPE normally shouldn't be used, you don't need a world, and you don't need to touch make. Just follow the existing models as a guide and edit models, rcs, and launch params approperately.

jannsta1, 2018-09-29:
@lamping7 - could you confirm the directory that a custom rcs file for a posix_sitl_default build should go (running with sitl/gazebo)? Have tried putting my custom file, 'bee', in ```Firmware/build/posix_sitl_default/tmp/rootfs/etc/init.d-posix``` but I get the following error: 

```
INFO  [Unknown] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Error: Unknown model 'bee'
etc/init.d-posix/rcS: 48: exit: Illegal number: -1
ERROR [Unknown] Startup script returned with return value: 512```

lamping7, 2018-09-29:
@jannsta1 it doesn't matter where you put it if you're just using a launch file to spin all this up as long as you set the change the parameters in the launch file to point there. You can also look at the supplied launch files to find the default location and put it there.

ChuplesKai, 2018-10-01:
I've also been trying to work through this and make it work, but I'm not having much luck.  In addition to the above instructions, I noticed that there's also a CMakeLists.txt in the configuration folder (Firmware/ROMFS/px4fmu_common/init.d).  Changing this has not helped get things running.

@lamping7 When you say that you need launch files to point to the model, which argument to the script are you referring to (the first argument or what comes after the -s)?

lamping7, 2018-10-02:
@bkueng Did some magic getting all this cleaned up, but it still essentially works the same, from what I can see (correct me if I'm wrong Beat, I haven't tried my custom models with master yet). Here are the main PRs: https://github.com/PX4/Firmware/pull/10173, https://github.com/PX4/Firmware/pull/10222; with some minor documentation changes here https://github.com/PX4/Devguide/pull/600. 

One new thing with launching the px4 app with a custom model is the `-s` for specifying the startup file. Before it was more of a positional arg. Anyway, just put the whole path and name of the startup file after the `-s`: https://github.com/PX4/Firmware/blob/181b2f69ed53d3c4d011f2b5465fac4c16e0ecd9/platforms/posix/src/main.cpp#L558 and https://github.com/PX4/Firmware/blob/181b2f69ed53d3c4d011f2b5465fac4c16e0ecd9/platforms/posix/src/main.cpp#L205

The cool thing that was added (that you normally won't be using with a custom model is the auto rcs loading): https://github.com/PX4/Firmware/blob/719bfd10730310f3b897dd7f6f0c44d24eb3d19a/ROMFS/px4fmu_common/init.d-posix/rcS#L42. So, by default it loads the generic startup and auto selects the right one according to the vehicle. None of this matters for your custom model though.

lamping7, 2018-10-02:
Additionally, in regards to "where to put the startup file" you can't just drop it in a folder anymore, like you used to be able to (my comment a few above this, may be misleading). You have to change the arg manually. Perhaps we should put the `rcS` arg back in the launch file? This will allow users to avoid having to directly modify the launch file. Any opinions here?

lamping7, 2018-10-02:
Yeah. That makes sense. But for now users can just maintain the whole rcS as how it previously was.

jannsta1, 2018-10-02:
I get some errors from a custom sdf file that previously worked OK - my sdf file is called ```bee``` but is pretty much the same as the old typhoon_h480 sdf. If I set  ```-s etc/init.d-posix/bee``` when launching the px4 node I get:

```
etc/init.d-posix/bee: 105: etc/init.d-posix/bee: vmount: not found
etc/init.d-posix/bee: 106: etc/init.d-posix/bee: camera_trigger: not found
etc/init.d-posix/bee: 108: etc/init.d-posix/bee: mavlink: not found
etc/init.d-posix/bee: 109: etc/init.d-posix/bee: replay: not found
```

lamping7, 2018-10-02:
You need to either revert back to 1.8.0 or fix your startup file according to the new way all this is arranged. Those modules are not in the same place. See the PRs I posted above.

bkueng, 2018-10-03:
@ChuplesKai What you are doing sounds correct, but I'd have to see all your changes to be able to diagnose it. 0001 should be fine. What are your file names and what vehicle names do you use?
The file in `init.d-posix` (you are doing it in-tree, right?) needs to be named `0001_<vehicle>`.

jannsta1, 2018-10-03:
OK, for a sanity check is ```$ make posix_sitl_default gazebo_typhoon_h480``` currently working without any warning messages? I see:

```
INFO  [Unknown] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Error: no autostart file found
etc/init.d-posix/rcS: 155: exit: Illegal number: -1
ERROR [Unknown] Startup script returned with return value: 512
```

I think this was working before (post the build changes discussed here) but I've tried completely wiping and re-cloning the firmware and all submodules and rebuilding and I still see this error.

bkueng, 2018-10-03:
Fix is up: https://github.com/PX4/Firmware/pull/10631

jannsta1, 2018-10-03:
Good news. Is it possible for someone to summarise all the steps required for a custom sitl config to be added to the new build structure? I've got:

- add gazebo model to Firmware/Tools/sitl_gazebo/models
- add config file to Firmware/ROMFS/px4fmu_common/init.d-posix 
(the file name must be of the form XXXX_MODEL_NAME, where XXXX is a unique number that can't begin with 0 and MODEL_NAME matches the gazebo model name)

This on its own doesn't seem to work. Is editing cmake or any other files required?

lamping7, 2018-10-03:
Don't forget to set the `vehicle` arg for the launch file, where `vehicle` matches what you called `MODEL_NAME`. This tells the launch file which model to load and sets the `PX4_SIM_MODEL` env variable so that the rcS can auto select and load the right the specific startup for that model. no cmake or other files is required. If someone wants to share their model and startup files, we can make an example of it.

jannsta1, 2018-10-03:
OK - had been trying using ```$ make posix_sitl_default gazebo_MODEL_NAME``` to test but will also try with the launch files.

No doubt there are better examples out there but in case its helpful my Frankenstein Iris/typhoon_h480 model is attached (iris body, typhoon gimbal and startup parameters altered for ekf2 to use mocap data). The rcS file is still in the old format. 

[bee_model_files.zip](https://github.com/PX4/Devguide/files/2442363/bee_model_files.zip)

ChuplesKai, 2018-10-03:
Thanks @bkueng!  That seems to have resolved issues and I'm finally seeing the script run to completion and my SDF loading in Gazebo!  For everyone's sake, I'll post what steps I took to see things running.

###   Getting your SDF to Gazebo

There are a couple of ways to do this it seems.  You can directly add your model to `Firmware/Tools/sitl_gazebo/models` as suggested by @jannsta1 which should be the simplest solution.  

Alternatively, you can have your SDF outside the Firmware tree by modifying your `GAZEBO_MODEL_PATH`environment variable to include an additional models directory.  This is what I did, and it seems to work fine.

In either case, make sure that you set the `sdf` argument of posix_sitl.launch appropriately to point to your model.

###  Building the Firmware for your model

I don't know if all of these steps are necessary, but this is everything I did (for FW 1.8):

- Add a custom configuration file for your platform under `ROMFS/px4fmu_common/init.d`, making sure the name is of the format ID_<vehicle>.  Note that as pointed out above, the ID must be unique and cannot start with leading 0's, as it is interpreted with those 0's stripped. (Perhaps there is some insight that could be given as to how these IDs should be set up?)
- If you want to perform SITL over your model, also add a configuration file under `ROMFS/px4fmu_common/init.d-posix`, ensuring to use the same ID and vehicle name.
- Modify `ROMFS/px4fmu_common/init.d/CMakeLists.txt` to include your vehicle in the list.
- Also modify `platforms/posix/cmake/sitl_target.cmake` to include your vehicle as outlined above (note that currently this appears to be at line 49).

jannsta1, 2018-10-03:
@ChuplesKai - thanks for sharing - could you post the git commit id that you're using? (I think some of these changes are post stable release v1.8.0). Appending GAZEBO_MODEL_PATH is a good idea too. From memory I think you can put stuff in ~/.gazebo/models and it's available without it being added to the path.

@lamping7 - A quick tutorial / example case would be really appreciated. Even if this was done with one of the existing airframes that is already in the new form (this is probably how most customised models probably start anyway) as a copy/paste exercise. I think the gazebo model part is fairly straightforward just the rcS file (if its still referred to as that) that's proving tricky.

ChuplesKai, 2018-10-03:
@jannsta1 I'm working off of commit `cfac2cc` (slightly out of date).  Now, as far as I know, this is an unstable development step, but it sounds like you have an idea that the process will be slightly different with the 1.8 stable release?  I'm still having some hard-to-identify issues here, so I'm wondering if I should be backing up to the stable release and working from that.

ChuplesKai, 2018-10-03:
Also, I can't seem to figure out what is going on, but somehow the custom model is causing the simulation to hang (I never see localization fusion starting).  This might just be a separate issue (so feel free to tell me so), but I think it might be related to an error I'm seeing early on in the initialization process: `INFO  [Unknown] Creating symlink /home/chuples/src/Firmware/ROMFS/px4fmu_common -> /home/chuples/.ros/etc
33 WARNING: setRealtimeSched failed (not run as root?)`

Now it seems weird to have to run as root, but could the lack of this symlink be causing PX4 to hang?

jannsta1, 2018-10-03:
@ChuplesKai - it seems like the Firmware build/boot process is undergoing some fundamental changes at the moment - some have been added after your commit level. This is welcome if you're using custom models as merging with Firmware updates is more straight-forward (as it won't be necessary to edit the cmake files) but its possible there are still some teething issues with the new setup - don't get too attached to your current setup just yet :). 

I'm not sure the rosinfo message you are seeing will cause a hang. Did you try setting the verbose flags in the px4 roslaunch files to true? I noticed that if gazebo can't find models it hangs, but you only get a warning if the verbose flag is set.

ChuplesKai, 2018-10-03:
@jannsta1 Right yeah, it seems things are still changing quite a lot... I'm going to need to try to reproduce all of this on stable because our company is working toward a deliverable, so I'll need something.. well, stable.

I haven't tried the verbose flag, but Gazebo does seem to load up fine and it shows my model.  Gazebo itself doesn't appear to throw any errors.

I realized I might need to do as you have and instead of invoking the generic build `make posix_sitl_default gazebo`, I should probably try the build with my model... but that's really not working (looks like it's an issue of needing a `<vehicle>.world` model, but it could also be other things.

ChuplesKai, 2018-10-04:
I'll try to make things more clear @lamping7.  There are a lot of reasons why this thing is confusing.

First of all this is the first time I've seen it recommended to run those two particular build targets.  It's very confusing when there are _so many different targets_.  Also note that those specific build targets have not resolved my issue.

Looking over your `.bashrc`, unless `PX4_SRC_DIR` is somehow used by the Firmware, my setup exports and sources the same stuff (as directed by the PX4 install instructions).

Now, your example of creating this IRIS copy doesn't quite fit for our needs with a custom SDF, I believe I have taken the necessary steps to create all of the appropriate configuration files, and I have a custom SDF file which does load in Gazebo as I would expect it to.

Despite doing the steps you've outlined, _it still doesn't work_.

lamping7, 2018-10-04:
@ChuplesKai Your not giving me enough info to help you. Saying it doesn't work doesn't help me help you. Regarding the build targets: There are many ways to do this. I just shared what I do and what is necessary to replicate the process I shared. If you carefully read the `.bashrc` you'll see that `PX4_SRC_DIR` is only used there. I use it because I often have different versions of the firmware I'm playing with and want to be able to quickly change things. You can ignore it if you like. @jannsta1 asked if I would do a simple example.
> Even if this was done with one of the existing airframes that is already in the new form (this is probably how most customised models probably start anyway) as a copy/paste exercise.

So I did as he requested. I would've used his model, but it doesn't adhere to the new configuration style and I don't have time to do it for him. Carefully look over the steps above and notice the changes that are required to match each other for this to work.

jannsta1, 2018-10-04:
Thanks @lamping7 - following these steps worked. I guess by the time I was adding the right files I was starting gazebo via make rather than roslaunch - this is why I was getting error messages. I'm not sure if this matters - I would normally use roslaunch so current level of support is OK for me. I tried following the same process for the typhoon_h480 (with associated .post file) and this also seemed to work. Could be worth adding something in the docos about these .post files?

ChuplesKai, 2018-10-05:
Going over the steps @lamping7 provided, I was able to recreate the working IRIS copy, even for the stable release of 1.8, which is definitely good progress.  There still seems to be some issue though when I try to use my custom model, which is increasingly leading me to believe that there is something in how my .sdf file is formatted which is upsetting the simulation.

Specifically, when running with my .sdf, the simulation seems to hang right after the spawn_model process of Gazebo finishes, whereas when I run the default IRIS model, I see it continuing on to start the simulation.  The following line:

`INFO  [simulator] Got initial simulation data, running sim..`

Appears when I run with IRIS or the IRIS copy, but not with my model.  The biggest notable difference is that my SDF is outside the Firmware tree, but I'm not sure why this would cause a problem.

jannsta1, 2018-10-06:
@ChuplesKai did you try running iris-copy out of tree?

ChuplesKai, 2018-10-08:
@jannsta1 good suggestion.  Trying it out, the iris-copy does indeed work out-of-tree.

Somehow, the custom .sdf file must be the issue.. it parses just fine in Gazebo but it somehow makes the simulator hang.

jannsta1, 2018-10-08:
sounds like either GAZEBO_RESOURCE_PATH or GAZEBO_MODEL_PATH are not fully populated or the XML tags in the .sdf have some issue. If you establish the paths are OK (just use $: ```echo $GAZEBO_[path of interest]```)   then you can just comment out sections of the .sdf until you find the offending section(s).

ChuplesKai, 2018-10-08:
Hey @jannsta1, I've been looking at things and poking at it a bit, but I'm pretty stumped.

As for the paths, I think things are set up correctly. The model path is pointing to my custom models directory, and I am able to see my model load in the Gazebo GUI.  There's no complaint from Gazebo at all, so I'm assuming that is okay.  You'll have to forgive my ignorance about the resource path, but I'm not sure that's a problem, as it seems to only really matter for loading .world files.  Now, I can see `iris.world` does exist in the Firmware tree, but I never created an `iris-clone.world` file, but the iris-clone still runs.  It just seems to default to `empty.world`, which is fine for now.

As for the .sdf model, I cannot figure out what might be wrong.  Doing as you suggested, I systematically began removing parts of the model to see if it would prevent the hang in the mavlink simulator, but it happens every time.  I've even tried stripping the model down to just a single base_link, but that still causes issues.  I thought the problem might involve having multiple collision/visual bodies on a single link, so I also removed all but one.. but that also does not seem to be the issue.  I almost think it's a problem that I've *omitted* something, but comparing my file with `iris.sdf`, I think I have everything that is relevant (including rotors and the imu link).

I've been able to track the hang to a specific loop of `simulator_mavlink.cpp` in FW 1.8 (Lines 746-780), but that's the best I've been able to figure out.  I haven't yet been able to decipher exactly what's going on in the code there.

lamping7, 2018-10-08:
Have you looked at the gazebo output when passing the verbose flag in the launch file?

ChuplesKai, 2018-10-09:
@lamping7 thanks for the advice!  I didn't think Gazebo was having trouble, but putting that verbose flag true reveals a very different story.  Hopefully this will let me fix whatever is upset.

S259420, 2018-11-10:
Hello, does anyone know how to set and launch px4 sitl using a user-defined world file? 
I'd like to use a custom world file but I do not know how to make posix it... I do not want to artificially modify an existing world (as I've done so far...) I didn't find any documentation, I would appreciate any suggestions.

hamishwillee, 2018-11-23:
@lamping7 you make a very good point where you say:

> I'm not really sure what is confusing you: I think it's the 10 month old info at the top.

@lamping7 @jannsta1 

We're at the point now that the current state of this needs to be a PR. That we can work through and test. Then future change and issues can be discussions against a clear baseline. Can one of you guys volunteer to create a PR based on our current knowledge? Then we'll ask the other and @ChuplesKai if they can test those instructions.

Looking at this. 
- We should target 1.8.1. 
- It should be created as a sub page of http://dev.px4.io/en/simulation/gazebo.html. I suspect we will later want to split it further, but first draft should be monolithic.
- We will miss some topics of interest in draft, like [designing the model](https://github.com/PX4/sitl_gazebo/issues/110#issuecomment-440628511). Put a placeholder in the structure for any interesting topic and for content have TBD.
- The structure in the very first post is a good start https://github.com/PX4/Devguide/issues/398#issuecomment-353411908 . 
- How likely is it that someone will start with an iris model and then build from that when they want to do a new model? (or to put it another way, is this a "good way" to create a model?)
  - Consider whether or not having a simple tutorial example at the end makes sense: https://github.com/PX4/Devguide/issues/398#issuecomment-426845908

> **NOTE** In general I am wary of tutorials. They are great if every step is very similar for all cases, but have a tendency to miss things that are off that centre path. So what I propose here is that whether or not we use a tutorial to illustrate, we start with the "these are the topics that we want to cover" as headings, and then start filling them in. @jannsta1 seems to have done that well but I'm not the right one to judge.

Thoughts?

hamishwillee, 2018-11-23:
PS. We know we want the structure to include topics related to modelling the motors.  @TSC21 proposed some topics needed: importing an URDF from SolidWorks and convert it to an SDF, what plugins to include in the model, modelling the motors, modelling the control surfaces ...

I am happy for many of these to be just TBD in the PR. But would be good if we could capture all the topic headings that we eventually intend to populate. So if there are any you would like to see, please add here.

hamishwillee, 2018-11-26:
@lamping7 @TSC21 - Jan does not feel confident to create the initial structure of topics. Would you be willing to do so and populate with the current steps? (As discussed above: https://github.com/PX4/Devguide/issues/398#issuecomment-441152892)

TSC21, 2018-11-26:
@hamishwillee Yeah I will do that along this week.

hamishwillee, 2018-12-03:
@TSC21  

> Yeah I will do that along this week.

Nag nag nag

hamishwillee, 2018-12-04:
@TSC21 That's fine :-) I'll keep nagging.

hamishwillee, 2018-12-12:
@TSC21 And here I am again :-)

RicardoM17, 2019-03-14:
Any progress here @hamishwillee ?? now it's my turn to nag :stuck_out_tongue_closed_eyes:

hamishwillee, 2019-03-14:
@RicardoM17 That's a question for @TSC21 :-) I'm not going to even attempt to progress this without him.

Oliver-ss, 2019-04-18:
> I'm not really sure what is confusing you @jannsta1 and @ChuplesKai: I think it's the 10 month old info at the top. I also don't think it is understood how the setup and launching works. The basic thing to know that might remove some confusion is that there are two main ways to launch PX4 and Gazebo. One is with `make` and the other is with `roslaunch`. I play around with ROS, so I'm going to only touch on that.
> 
> Let's do a simple example: I'm going to show you how to make a copy of the `iris` model and launch it, explaining it a bit on the way. I'll call this new vehicle `iris-copy`. Hopefully this will show you what parts are important, where they live, and what changes are required.
> 
> Prerequisites:
> 
>     1. Clone PX4 Firmware
> 
>     2. Enter PX4 Firmware directory
> 
>     3. Build the firmware
>        `make posix_sitl_default`
> 
>     4. Build sitl_gazebo, and do not launch gazebo yet
>        `make posix_sitl_default sitl_gazebo`
> 
>     5. Setup your environment. For example, this is in my `~/.bashrc`:
> 
> 
> ```
> # PX4
> export PX4_SRC_DIR=~/px4/Firmware
> source ${PX4_SRC_DIR}/Tools/setup_gazebo.bash ${PX4_SRC_DIR} ${PX4_SRC_DIR}/build/posix_sitl_default
> export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:${PX4_SRC_DIR}:${PX4_SRC_DIR}/Tools/sitl_gazebo
> ```
> 
> Here we go:
> 
>     1. Go to the models directory in your PX4 Firmware directory
>        `cd ~/Firmware/Tools/sitl_gazebo/models`
> 
>     2. Copy the model directory, and enter it
>        `cp -R iris iris-copy`
>        `cd iris-copy`
> 
>     3. Rename the file and some bits inside them
>        `mv iris.sdf iris-copy.sdf`
>        `sed -i 's/iris/iris-copy/g' model.config`
>        `sed -i 's/name='"'"'iris'"'"'/name='"'"'iris-copy'"'"'/g' iris-copy.sdf`
> 
>     4. Go to the startup (rcS) file directory
>        `cd ../../../../ROMFS/px4fmu_common/init.d-posix`
> 
>     5. Make a copy
>        `cp 10016_iris 10017_iris-copy`
> 
>     6. Enjoy!
>        `roslaunch px4 posix_sitl.launch vehicle:=iris-copy`

Hi, I just wondering if I want to use my own custom model, how could I create my model file after `cd ../../../../ROMFS/px4fmu_common/init.d-posix`

hamishwillee, 2019-05-02:
@TSC21 Would be nice to progress this ... 
I know you are always busy, so what can we do to make it more of a priority? If you will never have time available, is there someone else you can recommend to help with this?

imagotechteam, 2019-05-20:
We don't need to add a custom pixer to /ROMFS/px4fmu_common/mixer-sitl ?
It seems to be very welcoming for such file :)

imagotechteam, 2019-05-21:
> roslaunch px4 posix_sitl.launch vehicle:=iris-copy`
What would that be to launch a custom vehicle with make ?

hamishwillee, 2019-05-21:
@imagotechteam YOu post above is a technical support query, and not particularly relevant to this post. Please post on the discussion boards.

tuloski, 2019-05-27:
Can also someone explain how the .sdf are generated from the .xacro? For example for the iris model it seems that the .sdf are automagically generated from .xacro (inside the /models/rotor_description/urdf) which is much more readable.

jannsta1, 2019-05-27:
Did you read the gazebo help pages about URDF? e.g. http://gazebosim.org/tutorials/?tut=ros_urdf. I think it's explained quite comprehensively there. I haven't used them a lot but the way I think of it is that the URDF gets converted into the .sdf, xacros are a feature of URDF. e.g. https://answers.ros.org/question/202162/urdf-or-xacro/

[3 more replies]