PX4/PX4-Devguide #415: Gazebo Simulation - needs a section on sensor/GPS noise/delay

hamishwillee, 2018-01-03:
Bug Page: [Gazebo Simulation · PX4 Developer Guide](https://dev.px4.io/en/simulation/gazebo.html)

Should have a section explaining how to customise the GPS noise/delay as per https://github.com/PX4/Firmware/pull/8555#issuecomment-354979106. More generally, lets document all the important vehicle-specific settings in SDF and any other common environmental settings that people might want to change (wind?).

Probably group under a parent topic along with this heading: https://dev.px4.io/en/simulation/gazebo.html#set-custom-takeoff-location

ndepal, 2018-01-05:
This would be great.

The only SDF parameter that I'm aware of is the GPS noise:

https://github.com/PX4/sitl_gazebo/blob/1de7366d4e0cb7f784367d8669138d45781c3994/src/gazebo_gps_plugin.cpp#L54-L59

hamishwillee, 2018-01-08:
@ndepal Can you create some text? The reason I don't want to is that I really have no experience of gazebo. 

Let me outline my questions/concerns I'd want such a section to answer.
1. I understand that the SDF file is created as part of the PX4 build process? Doesn't that mean that it might get over-written every time you build PX4?
2. I did a build against current Firmware. I can find SDF files, but none seem to have gpsNoise in them. 
   - Is it not present by default? If it is, in which sdf files?
   - If it is not present by default that implies that gpsNoise is OFF by default?
   - Assuming we have to add this section in, what does it look like. i.e similar to this from file:
```
    <plugin name='rotors_gazebo_imu_plugin' filename='librotors_gazebo_imu_plugin.so'>
      <robotNamespace></robotNamespace>
      <linkName>plane/imu_link</linkName>
      <imuTopic>/imu</imuTopic>
      <gyroscopeNoiseDensity>0.0003394</gyroscopeNoiseDensity>
      <gyroscopeRandomWalk>3.8785e-05</gyroscopeRandomWalk>
      <gyroscopeBiasCorrelationTime>1000.0</gyroscopeBiasCorrelationTime>
      <gyroscopeTurnOnBiasSigma>0.0087</gyroscopeTurnOnBiasSigma>
      <accelerometerNoiseDensity>0.004</accelerometerNoiseDensity>
      <accelerometerRandomWalk>0.006</accelerometerRandomWalk>
      <accelerometerBiasCorrelationTime>300.0</accelerometerBiasCorrelationTime>
      <accelerometerTurnOnBiasSigma>0.196</accelerometerTurnOnBiasSigma>
    </plugin>
```
3. In example above we see some gyroscope noise, so does this not mean that there are other useful parameters people might want to set?
4. Is there anything we can say about the gpsNoise? For example, as a user I'd like to know what will happen if I set this - how much noise (debilitating?). When should I set this - i.e. to test what sort of conditions?

ndepal, 2018-01-08:
Lots of good questions, most of which I unfortunately can't answer.

1. I don't know anything about that either. It seems `iris` is generated, as there is no `.sdf` in its directory https://github.com/PX4/sitl_gazebo/tree/master/models/iris. Others don't seem to be generated, e.g. https://github.com/PX4/sitl_gazebo/tree/master/models/typhoon_h480.

2. I built `make posix_sitl_default gazebo_iris`. I find several models that have it enabled:
```./Tools/sitl_gazebo/models/hippocampus/hippocampus.sdf:691:        <gpsNoise>true</gpsNoise>
./Tools/sitl_gazebo/models/solo/solo.sdf:428:        <gpsNoise>true</gpsNoise>
./Tools/sitl_gazebo/models/iris/iris.sdf:427:      <gpsNoise>true</gpsNoise>
./Tools/sitl_gazebo/models/standard_vtol/standard_vtol.sdf:869:        <gpsNoise>true</gpsNoise>
./Tools/sitl_gazebo/models/delta_wing/delta_wing.sdf:257:        <gpsNoise>true</gpsNoise>
./Tools/sitl_gazebo/models/plane/plane.sdf:458:        <gpsNoise>true</gpsNoise>
./Tools/sitl_gazebo/models/typhoon_h480/typhoon_h480.sdf:1333:        <gpsNoise>true</gpsNoise>
./Tools/sitl_gazebo/models/tailsitter/tailsitter.sdf:655:        <gpsNoise>true</gpsNoise>
```
The absence of the parameter does mean GPS noise is off.

The parameter needs to be specified to the gps plugin:
```
<plugin name='gps_plugin' filename='libgazebo_gps_plugin.so'>
      <robotNamespace/>
      <gpsNoise>true</gpsNoise>
</plugin>
```

3. Most likely. I'd be interested in learning about others as well.

4. The GPS noise is not supposed to be debilitating. It's supposed to simulate somewhat realistic GPS measurements found in the real world. With GPS noise off the GPS measurements PX4 gets from Gazebo are completely noise free (i.e. perfect).
The noise parameters (of the decaying random walk used to simulate the GPS noise) are hard-coded here 
https://github.com/PX4/sitl_gazebo/blob/71a44b97ef4888eb2a247fbf67c79c65dd4e2ee0/include/gazebo_gps_plugin.h#L122. Unfortunately the HDOP/VDOP/EPH/EPV that this would correspond to are not published to PX4 (see https://github.com/PX4/sitl_gazebo/blob/71a44b97ef4888eb2a247fbf67c79c65dd4e2ee0/src/gazebo_gps_plugin.cpp#L143). When I added GPS noise (without an SDF parameter) I added a rough calculation that has since been removed (https://github.com/PX4/sitl_gazebo/commit/3e55f8f5c36b01c592ccab249ead13c52e01e887#diff-effb2388eb994641e3f8e86437253db2R169).
I was interested in this when integrating the precision landing camera with the state estimator. With perfect GPS measurements, I could not really observe an improved position hold performance thanks to the additional measurements. Adding noise to the GPS helped me do that.

This is about as much as I can contribute, I'm afraid, as I only just stumbled across the GPS noise SDF parameter.

hamishwillee, 2018-01-09:
@ndepal That is an excellent start, and actually about the right level (enough for me to understand why you would bother setting noise. I'm going to neglect the other settings ("3.") for now. Proposed new section is here:  https://github.com/PX4/Devguide/pull/426

hamishwillee, 2018-01-10:
Closed by #426

closed 2018-01-10 by hamishwillee