PX4/PX4-Devguide #419: Need tools to support/ease translation of docs from source

hamishwillee, 2018-01-04:
Currently parameters, airframes docs etc are produced from source. We don't have a way to support translation of these docs - [currently](https://github.com/PX4/Devguide/pull/414#issuecomment-355301369) we are just suggesting create another matching doc "may be out of date".

The goal of this is to look at ways to automate translation. What I think is that translations should be kept in separate files that can be injected into the build process. The system would use the translation if it exists, but otherwise use the English version. 

@dagar FYI I am happy to look at this in February, if you don't look at it sooner.

LorenzMeier, 2018-01-04:
I think the intent is good, but this doesn't seem to be first priority to me. I would propose to include the English pages and table this for later this year. We have so much to do on many other fronts and this could quickly become a time sink.

hamishwillee, 2018-01-04:
Agree. In the meantime I have proposed that if they want to translate they can do so, but rename that as a separate document, add a disclaimer about it possibly being out of date, and link to the English version.