PX4/PX4-Devguide #427: Add additional controller diagrams

hamishwillee, 2018-01-09:
As discussed in https://github.com/PX4/Firmware/issues/8316#issuecomment-354956186

@bresch and @Stifael volunteered!

bresch, 2018-01-31:
Here is the pitch controller (roll is the same but with _phi_ and _p_ instead of _theta_ and _q_)
[image]
@tubeme FYI

hamishwillee, 2018-02-01:
Thanks very much @bresch ! I hope to get these into draw.io next week.

hamishwillee, 2018-02-07:
@bresch Thank you for this. I've created a version on draw.io here you can edit further: https://drive.google.com/file/d/1ibxekmtc6Ljq60DvNMplgnnU-JOvKYLQ/view?usp=sharing 

[image]

A few statements/questions:
1. The subscript r is "required" I presume? I have changed this everywhere to d (desired) as that is consistent with other image. 
1. Everything here is a scalar right (ie I have removed all bold)
1. Please check I got it right. 
1. What "words" would you like with this - see https://dev.px4.io/en/flight_stack/controller_diagrams.html#multicopter-position-controller

Once you're sure this is OK I'll modify to create the roll controller.

@Stifael @bkueng Anything you would like to suggest/add for this?

bresch, 2018-02-07:
@hamishwillee Thank you for translating it into _draw.io_.
1. You did it right, I didn't respect the notation there!
2. Correct, however I'm thinking about including roll and yaw in the same diagram. It would also be nicer to have a vector going to the Jacobian block. I don't want to have mathematician trying to kill me xD
3. I will do the modifications and tell you when I'm done

hamishwillee, 2018-02-07:
Thanks very much. I hope you can also help with "4" :-)

bresch, 2018-02-16:
@hamishwillee I updated the schematic and it includes roll and yaw now.
[image]

hamishwillee, 2018-02-19:
Thanks @bresch . PR created https://github.com/PX4/Devguide/pull/464
This just needs some words from you and @Stifael - comments on inputs, outputs, anything you think is relevant and not obvious from the diagram.

tubeme, 2018-02-19:
Just something I cannot understand. I suppose this is the Pitch controller for airplane because there is airspeed measured.

If there is no airspeed how the diagram changes? What do we use instead of Vr?

bresch, 2018-02-19:
@tubeme When there is no pitot tube, the airspeed is estimated using the ground speed (but wind will degrade performance)
Check my comment here for more details on the controller: https://github.com/PX4/Devguide/pull/464#issuecomment-366632282

tubeme, 2018-02-20:
Thanks @bresch prior to 1.7 ie in 1.6.5 we had the option to choose a "not valid" for the FW_ARSP_MODE in this situation we neither use the ground speed nor the airspeed in the algorithm. Now in 1.7 there are only two options for the mode Air speed and ground speed. That is why I'm a little confused. If we choose not valid in 1.6.5 then where is this airspeed calculated from? this concerns the TECS.

bresch, 2018-02-21:
@tubeme I think I was wrong, in airspeedless mode, it appears that the ground speed is not used, gain scheduling for the FW attitude controller is then disabled and no correction is made in TECS using airspeed feedback (it's open loop).

hamishwillee, 2019-01-17:
@bresch This is all sorted by https://dev.px4.io/en/flight_stack/controller_diagrams.html#fixed-wing-attitude-controller and can be closed right?

hamishwillee, 2019-02-12:
This is now closed. There are other diagrams to do, but the ones discussed here are complete

closed 2019-02-12 by hamishwillee