PX4/PX4-Devguide #433: Doc Bug: HITL Simulation · PX4 Developer Guide

jingego, 2018-01-17:
DESCRIBE PROBLEM WITH DOCS HERE

Bug Page: [HITL Simulation · PX4 Developer Guide](https://dev.px4.io/en/simulation/hitl.html)

The documentation is hard to understand.  
as described in HITL Simulation Environment, the application  "simulator" on hardware will handle the mavlink messages and publish in relevant topics.  In PX4 configuration section, the autopilot board flashed normal Firmware and then selected HITL airframe. For either px4fmu-v2_default or px4fmu-v3_default, the 'modules/simulator' was not built. how could this HITL structure work?
The main problem is that I cannot understand this HITL structure. Could anyone kindly help to descript it in detail?
I have not successfully run this HITL simulation since I cannot see the same thing in step 2 of PX4 configuration. So I simply set the SYS_HIL parameter to enable. I may make some mistakes. I will try it again.

jingego, 2018-01-18:
[image]

I further read the code, it should be [mavlink_receiver.cpp](https://github.com/PX4/Firmware/blob/85ba1607575575340a4d4e02f7a2176475255d44/src/modules/mavlink/mavlink_receiver.cpp#L1828-L1965) will handle the HIL message. Note (Line 33) in [ Simulation](https://github.com/PX4/Devguide/blob/645f103e0b6989da44676be848d8966cd5f5840d/en/simulation/README.md#L33) may better to revise for better understanding.

hamishwillee, 2018-01-31:
@jingego Thanks very much for posting this. I *think* you are right. I also agree the solution is to change the diagram above to say mavlink_reciever.cpp rather than simulator_mavlink.cpp.

@dagar As per this post, my *new* understanding is that the simulator API is implemented in  **simulator_mavlink.cpp** for SITL. For HIL it is implemented in [mavlink_receiver.cpp](https://github.com/PX4/Firmware/blob/85ba1607575575340a4d4e02f7a2176475255d44/src/modules/mavlink/mavlink_receiver.cpp#L1828-L1965). The mavlink receiver code is only executed in HIL mode and the simulator_mavlink isn't present in firmware, so there is never the case where there are two handlers present for the simulation API messages.

Upshot, the HIL diagram needs to be changed to say mavlink_reciever.cpp. Does that make sense to you?

hamishwillee, 2018-01-31:
This will be fixed by https://github.com/PX4/Devguide/pull/443

jingego, 2018-03-06:
@hamishwillee many thanks！I just have another small question. The gazebo simulation supports many airframe at this moment, including fixed-wing, vtol etc. Will HITL simulations of these airframes be supported?

hamishwillee, 2018-03-06:
> @hamishwillee many thanks！I just have another small question. The gazebo simulation supports many airframe at this moment, including fixed-wing, vtol etc. Will HITL simulations of these airframes be supported?

@jingego Probably eventually - I think that we're settling towards gazebo as the main HITL platform. @dagar Do you know of any specific work to support other airframes - specifically the "generic" ones?

closed 2018-03-06 by hamishwillee

dagar, 2018-03-30:
No not actively, but it's certainly an idea I'd like to encourage.