PX4/PX4-Devguide #435: Adding new air-frame for multicopter.

chuong98, 2018-01-19:
We want to do an experiment with a new configuration, as illustrated in the Fig 
[image]
The idea is: 
M1 is the big motor at the Center of the Gravity to create 99% thrust, and facing down.
M2 is small motor to control positive pitch, facing up
M3,M4 are the small motor, M2 tilts 45deg, and M4 tilts -45deg, to create yaw moment to balance with the big motor, while also control roll and pitch. 

Take the center motor at the origin, and follow NED coordinate, the position of motor is
#motor  (x,y,z)         (direction in (x,y,z) )
M1        (0,0,0)              (0,0,1)
M2       (-R,0,0)             (0,0,-1)
M3       (-R/2,R,0)   (cos(45),0, -cos(45))
M4       (-R/2,R,0)   (-cos(45), 0,-cos(45))

I would like to create quad_cT config in the mixer file, how should I do it. Thanks for your help.

dagar, 2018-01-19:
@jlecoeur FYI

jlecoeur, 2018-01-19:
@chuong98 cool design! If I may suggest something, M3 and M4 will create more pitch moment than M2 so you should either place M3 and M4 closer to the CG, or tilt them at 60° instead of 45°. 

## Step 1 Geometry
### 1-a Geometry file
Add a geometry file in the folder `src/lib/mixer/geometries/` named **quad_ct.toml** with the following content:
```toml
# Experimental multirotor

[info]
key = "4ct"
description = "Experimental multirotor"

[rotor_default]
direction = "CW"
axis      = [0.0, 0.0, -1.0]
Ct        = 1.0
Cm        = 0.05

[[rotors]]
name      = "M1"
position  = [0.0, 0.0, 0.0]
Ct = 100
Cm = 5

[[rotors]]
name      = "M2"
position  = [-1.0, 0.0, 0.0]
direction = "CCW"

[[rotors]]
name     = "M3"
position = [0.5, -1.0, 0.0]
axis = [0.7071, 0.0, -0.7071]
direction = "CCW"

[[rotors]]
name     = "M4"
position = [0.5, 1.0, 0.0]
axis = [-0.7071, 0.0, -0.7071]
direction = "CCW"
```
Ct are the  thrust coefficients and Cm are moment coefficients, I put arbitrarily larger values for M1 but you will have to adjust it to your motors. For this you can measure the thrust ratio between M1 and M2 and use the same ratio for the respective Ct-s. Same thing for the moment coefficients.

### 1-b Cmakelists
Add quad_ct.toml to the geometry list here:
https://github.com/PX4/Firmware/blob/master/src/lib/mixer/CMakeLists.txt#L45

jlecoeur, 2018-01-19:
## Step 2 Mixer file
Add a mixer file in the folder `ROMFS/px4fmu_common/mixers/` named **quad_ct.main.mix**
You can follow the existing guide: https://github.com/PX4/Devguide/blob/master/en/airframes/adding_a_new_frame.md
Your mixer file will be similar to **quad_x.main.mix**, just replace `4x` by `4ct`.

## Step 3 Airframe file
Add an Airframe file in the folder `ROMFS/px4fmu_common/init.d/`
Again you can follow the devguide (link above)

## Step 4 Send us the video!

chuong98, 2018-01-20:
Wonderful. I will make it and update the status. Thanks so much for your super fast reply.

chuong98, 2018-01-20:
So, I followed your instructions, but there are two points I want to confirm:
1. SInce the M1 facing down, should I add the axis [0,0,1] as follow.
```
 [[rotors]]
name      = "M1"
position  = [0.0, 0.0, 0.0]
axis = [0.0, 0.0, 1.0]
Ct = 100
Cm = 5
```
2. How is the name 4ct created. Should I make any changes in the file multi_table.py? How does it know that 4ct will link to quad_cT.toml.
Thanks for your suggestion. Indeed, we placed M2,M3 about R/4, but I just gave an example R/2 to match with the fig. Thank so much for your help again.

chuong98, 2018-01-20:
Ok. So, if I make the name is quad_ct.toml (instead of quad_cT.toml) then every thing is compiled correctly. It works now. Thanks again.

jlecoeur, 2018-01-20:
No problem, let me know how it goes!
1. No, what matters is the thrust axis, as it is pushing upwards then the axis is [0.0, 0.0 -1.0]
2. `4ct` is the key that you provide in the info section of **quad_ct.toml**. 
**mixer_table.py** is not used anymore (it was even removed in master),  the new script is **px_generate_mixers.py**.
This script does not have to be changed, it knows **quad_ct.toml** because you added it in the CMakelists.
3. You can rename **quad_ct.toml** to **quad_cT.toml** if you wish (and update the CMakelists accordingly). Also you will see that the geometry file, mixer file and airframe file do not need to be named the same. It could be **quad_ct_geom.toml**, **quad_ct_mixer.main.mix** and **12345_quad_ct_airframe**.

hamishwillee, 2018-01-30:
@jlecoeur As a result of this discussion, is there anything that needs to be added to to (that is missing from) the devguide?
Brief scan seems to indicate that we don't have anything about the geometries file, is this something you could add to the instructions for building an airframe?

chuong98, 2018-01-30:
The tutorial in the previous discussion is complete to add new configuration. no additional information is needed.

hamishwillee, 2018-01-30:
@chuong98 Thanks. I still need to make sure that the Devguide captures all of this - hence question for @jlecoeur

hamishwillee, 2018-01-30:
@jlecoeur Ignore my comment above, I see you have an action to do this update in https://github.com/PX4/Devguide/issues/349 - any ETA on this?

jlecoeur, 2018-04-12:
@chuong98 were your tests successful? 
Your platform covers many features of the mixer generator and toml files. It would make a perfect tutorial in the documentation. Are you able to share pictures of the platform, and a video of the flight?

chuong98, 2018-04-12:
@jlecoeur Hi Jlecoeure, I already graduated and my team is still working on it. We have a competition of designing a new prototype for sUAV. At that moment, we can not release much the information, but here are  the pictures  of the first prototype before I left.

[image]

hlal1, 2018-04-16:
Hi Jlecoeure, I'm working on this sUAV prototype. I found out the moment and thrust coefficients for M1 which leads to yaw equilibrium. But, the sUAV seems to pitch forward and flip over everytime I increase the throttle. Could you let me know how I can incorporate this into my mixer geometry.

tzehwei, 2018-04-26:
@jlecoeur ,

Thank you so much for your explanation, it has actually giving us a good picture on how to add a new frame. 
The script is quite straight forward, however besides the [rotor_default] part.
Could you please elaborate each of the code line, such as the reason and meaning as well as the logic to find the value of Ct&Cm.

Appreciated your reply.
Thanks again!

hamishwillee, 2018-12-12:
@jlecoeur This has been sitting around for a while, can you please give me an updated ETA? If we can't do a proper job, would it not be useful to create a placeholder linking to all these open issues on the topic:
* [Adding new air-frame for multicopter. #435](https://github.com/PX4/Devguide/issues/435)
* [Add documentation on multicopter mixer #349](https://github.com/PX4/Devguide/issues/349)
* [Quadcopter with asymmetrical/different arms length #511](https://github.com/PX4/Devguide/issues/511)

Are there any other resources we might add?

In addition, could you advise on http://discuss.px4.io/t/mixer-and-geometry-for-new-vehicle-documentation-and-help/8843 ?

catchrow1, 2020-04-27:
The April 2020 call for a dev guide on this topic please as I/we too, have a unique config we would like to program in :-)

hamishwillee, 2020-04-28:
@jlecoeur @bkueng FYI, reminder. Beat, at what point do we give up on Julien? For myself I do not have time to commit to this (even to help) for about 2 weeks.

bkueng, 2020-04-28:
I currently don't have time either. Can you ping we when you have time again?

hamishwillee, 2020-04-29:
I'll try, but the way it is looking right now, "not soon".

catchrow1, 2020-04-30:
@chuong98   can you please upload your cmakelist.txt file?  Im getting errors at this point and want to see what it should look like!
Thanks!

JARC99, 2020-11-20:
@ryanjAA would it ber possible to share with me the jupyter notebook, I am myself trying to add a new config but I'm having trouble with the scattered information online.

ryanjAA, 2020-11-24:
@JARC99 it is here: https://github.com/jlecoeur/servo_mix_matrix

This lets you pretty much do anything needed. Just takes a bit of work to go through not the notebook but the whole TOML process for recompiling, it does however work fine once you spend a little time getting to know what data is wanted, etc.