PX4/PX4-Devguide #436: Doc Bug: HITL Simulation · PX4 Developer Guide

ndepal, 2018-01-22:
Gazebo HITL is not actually supported yet. Open PR here: https://github.com/PX4/sitl_gazebo/pull/157
I just spent about an hour trying to get this working according to the docs. They should mention that this is not in master yet.

Bug Page: [HITL Simulation · PX4 Developer Guide](https://dev.px4.io/en/simulation/hitl.html#simulator-setup)

TSC21, 2018-01-22:
Merged on `sitl_gazebo` side. TODO: add support for other airframes (on PX4 side).

closed 2018-01-22 by TSC21

ndepal, 2018-01-23:
Thanks. It's still not available in Firmware master though.

LorenzMeier, 2018-01-25:
@TSC21 Could you send a PR?

TSC21, 2018-01-25:
It's already on master.

ndepal, 2018-01-25:
I see that it was added yesterday https://github.com/PX4/Firmware/pull/8752. Thanks!

DanielPalominoS, 2019-08-23:
Hello, I have been working with a pixhawk 2 in HITL. When I use either 1.7.x or 1.8.x firmware version sensor messages are received. However, when I use 1.9.x firmware version this is not the case, same as happened before to @arwagoner  in [PX4/sitl_gazebo#157](https://github.com/PX4/sitl_gazebo/pull/157). I wonder if  this is any issue associated with this specific board or perhaps is a general behavior in 1.9.x firmware upgrade.
Sorry if this is not the right place to post, I was not sure where to do it. 
Thank you in advance to those who could provide any help.

TSC21, 2019-08-23:
@DanielPalominoS please bring your questions to PX4 Slack or to discuss.px4.io