PX4/PX4-Devguide #440: Gazebo Simulation - Add docs on Jinja2 based models

hamishwillee, 2018-01-29:
From https://github.com/PX4/sitl_gazebo/pull/37#issuecomment-358798665

Documentation: (@hamishwillee FYI):
* Should be documented how to create jinja description files that the jinja_gen reads and parses to .sdf;
* Comparison between jinja based vs xacro based generation (why? performance? easy of use?)
* Add the new airframe (the DJI Matrice 100) to the devguide - https://dev.px4.io/en/simulation/gazebo.html

hamishwillee, 2018-01-29:
@jgoppert @TSC21 Wasn't clear to me from devguide if you plan to do this or if you expect me to. Can you confirm? I can take this on, but if so may need to talk to you guys.

TSC21, 2018-01-29:
@hamishwillee you can easily add the new airframe. The questions regarding to the use of jinja over xacro should be discussed first.

hamishwillee, 2018-01-30:
@TSC21 Ah, so the case here is that we now provide **both** jinja and xacro for generating models, so we need to explain when/why you use each?

As above, "Wasn't clear to me from devguide if you plan to do this or if you expect me to. Can you confirm? " 
What I'd like is one of you to volunteer and me to then subedit (as that will be faster). However if you want me to take first shot please ask explicitly. What I want to avoid is me taking this on at the same time as you or @jgoppert

hamishwillee, 2018-02-12:
@TSC21 @jgoppert Can you please comment on https://github.com/PX4/Devguide/issues/440#issuecomment-361461334

TSC21, 2018-02-12:
> @TSC21 Ah, so the case here is that we now provide both jinja and xacro for generating models, so we need to explain when/why you use each?

Yeah, we can say that.

> What I'd like is one of you to volunteer and me to then subedit (as that will be faster). However if you want me to take first shot please ask explicitly. What I want to avoid is me taking this on at the same time as you or @jgoppert

You probably want to take a first shot if you don't mind, as me and @jgoppert are somewhat busy atm. I don't mind review it after that.