PX4/PX4-Devguide #454: Doc Bug: Gazebo Simulation · PX4 Developer Guide

hamishwillee, 2018-02-07:
Bug Page: [Gazebo Simulation > Starting Gazebo and PX4 Separately](https://dev.px4.io/en/simulation/gazebo.html#start_px4_sim_separately) doesn't work.

Anecdotally described in https://github.com/PX4/sitl_gazebo/issues/113 along with a solution. 

Need to determine how that fails and whether the suggested solution is an adequate replacement. If it does fail, need to explain "how" it doesn't work.

hamishwillee, 2018-02-07:
Questions:
1. Can Gazebo work in VM. I've had it working in past, but now it always crashes, making testing difficult.
2.  https://dev.px4.io/en/simulation/gazebo.html#start_px4_sim_separately suggests you use `make posix_sitl_default gazebo_none_ide` but `gazebo_none_ide` does not exist. Is `gazebo___ide` the same thing?
3. That doc also says "IDE select px4_<mode> target you want to debug (e.g. px4_iris)". Are there instructions to import a project into an IDE such that you can see these targets? I s there a project file for QtCreator?
4. HOw does `make posix_sitl_default gazebo no_sim=1` differ from using an ide target - ie `make posix_sitl_default gazebo___ide`? If you use the no_sim option can you still debug? (edited)

hamishwillee, 2018-02-07:
@TSC21 Any advice on this one?

dagar, 2018-02-07:
Maybe we should start with the basics and build up?

How would you start gazebo manually without multiple layers of scripting?
 
 - set the environment varliabe `GAZEBO_MODEL_PATH=Tools/sitl_gazebo/models`
 - run gazebo with the desired world

    export GAZEBO_MODEL_PATH=Tools/sitl_gazebo/models
    gazebo Tools/sitl_gazebo/worlds/iris.world

hamishwillee, 2018-02-07:
@dagar The linked doc suggests
```
cd workspace/Firmware
export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:build/posix_sitl_default/build_gazebo
gazebo Tools/sitl_gazebo/worlds/iris.world --verbose
```
How are the GAZEBO_MODEL_PATH and GAZEBO_PLUGIN_PATH related? Ie why are the solutions different?

Then in separate window run: 
```
make posix_sitl_default gazebo no_sim=1
```

I'm very happy to have whatever is correct here as a solution - and agree starting with basics makes sense. But would also like to get the IDE thing working. So perhaps that is a separate topic.