PX4/PX4-Devguide #458: OctoMap does not work with ROS Kinetic

tomatac, 2018-02-11:
OctoMap does not work with ROS Kinetic
When running:
roslaunch octomap_server octomap_tracking_server.launch I get a warning:
[ WARN] [1518384924.265631993, 49.000000000]: Nothing to publish, octree is empty
the Gazebo simulator stops working

gzserver: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreRenderSystem.cpp:546: virtual void Ogre::RenderSystem::setDepthBufferFor(Ogre::RenderTarget*): Assertion `bAttached && "A new DepthBuffer for a RenderTarget was created, but after creation" "it says it's incompatible with that RT"' failed.
Aborted (core dumped)
[gazebo-2] process has died [pid 42332, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -u -e ode /home/toma/catkin_ws/src/rotors_simulator/rotors_gazebo/worlds/basic.world __name:=gazebo __log:=/home/toma/.ros/log/4722d018-0f73-11e8-8b8e-000c296f4c62/gazebo-2.log].
log file: /home/toma/.ros/log/4722d018-0f73-11e8-8b8e-000c296f4c62/gazebo-2*.log
There is no Gazebo log file in the folder above

Ref: Bug Page: [OctoMap Models with ROS · PX4 Developer Guide](https://dev.px4.io/en/simulation/gazebo_octomap.html)

hamishwillee, 2018-02-12:
@TSC21 Is this something you could look at? (or recommend someone to check out)?

TSC21, 2018-02-12:
@tomatac are you running this on a VM? This is not related to Octomap, but rather with Gazebo itself. There are other users out there which reported the same problem. The source of the problem is reported here: https://bitbucket.org/osrf/gazebo/issues/1837/vmware-rendering-z-ordering-appears-random. This may also be related with your dedicated graphics card, so issue like `export SVGA_VGPU10=0` on the terminal before you launch the node, or you may want to try to re-installed the Nvidia/AMD proprietary driver.

tomatac, 2018-02-13:
@TSC21 I am using a VMware VM and I have the "export SVGA_VGPU10=0"
The graphics card is virtualized and the 3D acceleration is on.

Gazebo works great when I run: 
"roslaunch rotors_gazebo mav_hovering_example.launch mav_name:=firefly world_name:=basic"

When I run "roslaunch octomap_server octomap_tracking_server.launch" Gazebo crashes.

I would appreciate any suggestion for troubleshooting.

TSC21, 2018-02-13:
The solutions that I gave are the ones that I founs people have used to fix it. I don't have any other solution atm.

tomatac, 2018-02-13:
@TSC21 Thank you for your input! From the post you suggested to look at, I concluded that I should try to upgrade Gazebo from 7.0 to 7.5. I am not quite sure if that breaks ROS Kinetic compatibility and what are the best practices to upgrade. I am researching that and I will report back the results. If have any info for that feel free to post. Thanks!

TSC21, 2018-02-14:
I didn't said to update the Gazebo version but rather your graphics card drivers.

tomatac, 2018-02-14:
I am running in a VmWare virtual machine. I am not sure how to upgrade the graphics card drivers. They are all virtualized and there are not too many choices.
In the link you included above somebody mentioned that Gazebo 7.5 fixed some of the graphics issues. I upgraded Gazebo from "7.0.0+dfsg-2" to "7.11.0-1~xenial" (latest version) and I do not get the error anymore.
I am not sure if this upgrade will still be compatible with ROS Kinetic.

hamishwillee, 2018-02-23:
@TSC21 Does anything have to happen in Devguide for this? It is a bit of a challenge because ROS uses its own standardised Gazebo 7, and VMs are not really a supported plaform for us. 

I am tempted to close this and do nothing. Search should discover this issue and the possible solution for anyone getting the error.

TSC21, 2018-02-23:
People should adapt the dependency versions according to the platform they are using and any problem they come up with usually have its solution available after some search (as I did). So yeah let's close this.

closed 2018-02-23 by TSC21

nkhedekar, 2018-05-19:
The issue is solved by installing gazebo7 (version 7.12) as is given here - http://gazebosim.org/tutorials?tut=install_ubuntu