PX4/PX4-Devguide #470: Flight Mode Evaluation Diagram needs to be updated

barzanisar, 2018-02-23:
I discussed with @bkueng, there are two issues with this diagram in  https://dev.px4.io/en/concept/flight_modes.html :

1) The parameters OFFBOARD_SW and OFFBOARD_TH etc need more explanation. @bkueng says "*_TH" relate to the values we set in Multi Channel Mode Selection by sliding the threshold bars and "*_SW" is the value we set by switching the RC switches to select the flight modes. Also, I think these parameters set the value of the parameters “RC_MAP_OFFB_SW” and “RC_OFFB_TH” in QGC. Same goes for the rest of the parameters. Is this true @bkueng ?

I think explaining what these parameters are or providing a link to their explanation will make the diagram easier to understand.

2) The diagram needs update. For example, according to @bkueng, the fixed wing also has ACRO mode so the parts enclosed in red rectangles should be removed:  
[image]

bkueng, 2018-02-23:
> Is this true @bkueng ?

Yes. I suggest you test it out. Can you make the necessary changes?

hamishwillee, 2018-02-24:
Just adding a note from @bkueng : https://github.com/PX4/Devguide/pull/457#issuecomment-367942127

"This is for the (old) hierarchical mode switching. It's still relevant, but I think the acro needs to be updated. It should also mention that by default (on the QGC side, which might be documented already), a single mode-switch (usually with 3 states) is used. @RomanBapst can help with that as well."

hamishwillee, 2018-02-24:
> I think explaining what these parameters are or providing a link to their explanation will make the diagram easier to understand.

Yes please. I suggest a link, possibly in a table at the end.

barzanisar, 2018-03-01:
> Yes please. I suggest a link, possibly in a table at the end.

I couldn't find a link to the explanation of these parameters in px4dev guide. Should I add an explanation of these parameters in a paragraph or table?

barzanisar, 2018-03-01:
> It should also mention that by default (on the QGC side, which might be documented already), a single mode-switch (usually with 3 states) is used. @RomanBapst can help with that as well.

@bkueng @RomanBapst  I didn't understand this point. Why is it important to mention this in flight mode evaluation diagram?

hamishwillee, 2018-03-01:
>> Yes please. I suggest a link, possibly in a table at the end.

> I couldn't find a link to the explanation of these parameters in px4dev guide. 

All parameters are documented in https://dev.px4.io/en/advanced/parameter_reference.html - so for example https://dev.px4.io/en/advanced/parameter_reference.html#RC_MAP_OFFB_SW

IF there are parameters you can't find in that doc then they may have been renamed, and we go back to Beat to find their equivalents.

> Should I add an explanation of these parameters in a paragraph or table?

Yes. What I normally do is something like the table [here](https://docs.px4.io/en/flight_modes/takeoff.html#multi-copter-mc). It shows the relevant parameters (linked to official definition) with a very succinct explanation.

barzanisar, 2018-03-05:
I think "OFFBOARD_SW" is the PWM value set for the offboard mode which is assigned on the channel set in “RC_MAP_OFFB_SW”.
But I can't find the parameter that exactly defines "OFFBOARD_SW" in https://dev.px4.io/en/advanced/parameter_reference.html  
@bkueng said something about state machine parameters that define "OFFBOARD_SW" value but are not in the parameter reference list. @bkueng can you please confirm?

bkueng, 2018-03-05:
`OFFBOARD_SW` is just the current offboard switch state (in the range of [0,1]), that is sent from the RC. `RC_MAP_OFFB_SW` defines which RC switch/channel is mapped to control offboard.

barzanisar, 2018-03-16:
According to @bkueng this hierarchical flight mode switching will be removed and replaced in the near future. I guess then explaining this diagram will go in vain?

bkueng, 2018-03-16:
> According to @bkueng this hierarchical flight mode switching will be removed

This has not been decided yet. But yes, I'd wait until we have clearance on that. The discussion is here for reference: https://github.com/PX4/Firmware/pull/8999