PX4/PX4-Devguide #476: uORB graph: broken links for multi-topics

jlecoeur, 2018-03-07:
On [uORB Graph](https://dev.px4.io/en/middleware/uorb_graph.html), you are redirected to the topic definition when double clicking on a topic. For example, a double-click on `vehicle_rates_setpoint` redirects to [vehicle_rates_setpoint.msg](https://github.com/PX4/Firmware/blob/master/msg/vehicle_rates_setpoint.msg). That is really helpful.

However this does not work with multi topics.

For example, a double-click on `mc_virtual_rates_setpoint` redirects to
[mc_virtual_rates_setpoint.msg](https://github.com/PX4/Firmware/blob/master/msg/mc_virtual_rates_setpoint.msg) (which does not exists), when it should redirect to [vehicle_rates_setpoint.msg](https://github.com/PX4/Firmware/blob/master/msg/vehicle_rates_setpoint.msg) because `mc_virtual_rates_setpoint` has the same message definition as `vehicle_rates_setpoint`.

FYI @bkueng 
Should metadata be generated along with uORB code (or the generated uORB code be more explicit) when a topic is actually a multi topic, so that it can be detected when generating the graph?

bkueng, 2018-03-07:
There's a TODO for that here: https://github.com/PX4/Firmware/blob/master/Tools/uorb_graph/create.py#L566.
The difficulty is that the graph script does not know anything about the `.msg` files, and does not depend on any output of the generator script, it just extracts the topic names from source. To know about multi-topics, the script will need this additional information from the message files in one way or another. I haven't looked into that, what the best solution would be.

jlecoeur, 2018-03-07:
Although it would require renaming most existing multi-topics, I think enforcing a specific format for multi-topic names would be the simplest solution. And could also enhance readability of the code ("where can I find the message definition for this topic?")

Something like:
`name_of_topic`__`postfix`

example:
`vehicle_rates_setpoint`
`vehicle_rates_setpoint__virtual_mc`