PX4/PX4-Devguide #477: uORB: naming of multi-topics vs multi-instance

jlecoeur, 2018-03-07:
The [uORB dev guide](https://dev.px4.io/en/middleware/uorb.html) is a bit confusing now because it calls **instance** both 
- the multiple topic names defined with `# TOPICS` in .msg files
- the multiple topic instances advertised with `orb_advertise_multi`

The [uORB generation script](https://github.com/PX4/Firmware/blob/master/msg/tools/px_generate_uorb_topic_files.py#L92) calls the former **multi-topics**.

What is the correct naming convention?

bkueng, 2018-03-07:
Multi-instance (or just instances) is for `orb_advertise_multi`, and usually `multi-topic` is used for multiple topic names (via `# TOPICS`).
Do you want to update the docs accordingly?

jlecoeur, 2018-03-07:
Thanks! Here is the (tiny!) PR #478

bkueng, 2018-03-07:
Tiny indeed! Thanks :)

PR #478 (2018-03-07): uORB: do not call multi-topics 'instances'
closes #477
commit 343aa6cd (2018-03-07): uORB: do not call multi-topics 'instances'

closes #477
closed 2018-03-07 by bkueng