PX4/PX4-Devguide #508: (potentially) Error note

BazookaJoe1900, 2018-04-15:
@dagar 
I am not 100% sure if I am right or not but I think that there is an error here:
https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html?q=#what-should-i-do-if-the-height-estimate-is-diverging
(I think that)  the erorr is _"ekf2_innovations.vel_pos_innov[3] and ekf2_innovations.vel_pos_innov[5] will both have the same sign."_
but vel_pos_innov[3] should be vel_pos_innov[2]. that will make more sense because [2] and [5] are both relevant to Z axis.

hamishwillee, 2018-04-15:
@priseborough This one for you

BazookaJoe1900, 2018-12-17:
@hamishwillee @priseborough, is my note true?
I can do the pr if it is...

hamishwillee, 2018-12-17:
@BazookaJoe1900 I have no idea, which is why I deferred to @priseborough . Let's broaden the search - @dagar @mhkabir Do you know?

priseborough, 2018-12-18:
Yes, should be vel_pos_innov[2]

hamishwillee, 2018-12-18:
THanks @priseborough 

@BazookaJoe1900 Yes. adding a note as a PR would be great. Please link to here.

closed 2018-12-21 by BazookaJoe1900